yaapu/FrskyTelemetryScript #117: No telemetry on RadioMaster TX16s

ramiss, 2021-06-17:
Hi There.  I found your script through Painless 360s videos and cannot get telemetry to show on Yaapu. I have iNav widget right next to it, which shows telemetry just fine.  I'm using Crossfire and Ardupilot over Mavlink.  In fact, its the exact same scenario as the issue posted at the link below:

https://gitmemory.com/issue/yaapu/FrskyTelemetryScript/76/713585378

I did try re-discovering sensors while the script was loaded (and I assume running).  

Any help is appreciated.

yaapu, 2021-06-18:
My widget requires CRSF as RC protocol, with mavlink, which creates a fake CRSF telemetry emulation, it will not work.

closed 2021-06-18 by yaapu

ramiss, 2021-06-18:
Interesting, but Ardupilot does not officially support crossfire protocol yet.  So are you saying I have to use the development version?

yaapu, 2021-06-18:
yes, but you can go with the betas which have been released recentely