yaapu/FrskyTelemetryScript #120: Add support for internal GPS (draw it on the MAP)

yaapu, 2021-07-04:


AlksSAV, 2021-07-05:
Do you want to use the internal GPS that is in the radio equipment? I can test, I use betian 800 GPS in Radiomaster tx16s.

yaapu, 2021-07-05:
yes this is the idea, nothing to test yet, but thank you!

FasterRider, 2021-07-05:
> I use betian 800 GPS in Radiomaster tx16s.

A while ago I was looking for info on how to add a GPS module to TX16S and how to make use of it, but couldn't find anything. Do you have a link maybe?

yaapu, 2021-07-05:
https://github.com/opentx/opentx/pull/7632

FasterRider, 2021-07-05:
Thanks, I will add a GPS to the TX16S as soon as your new feature is available. Seeing both the craft's and and the transmitter's position on the map could be really helpful, especially while locating a crashed craft.

I just installed the latest version - looks like the maps are now integrated, making the separate mapping widget obsolete? (I haven't had time yet to check configuration more closely.)

yaapu, 2021-07-06:
> I just installed the latest version - looks like the maps are now integrated, making the separate mapping widget obsolete? (I haven't had time yet to check configuration more closely.)

The map has been there for a long time, the mapping widget is targeted at non ArduPilot users

FasterRider, 2021-07-06:
> > I just installed the latest version - looks like the maps are now integrated, making the separate mapping widget obsolete? (I haven't had time yet to check configuration more closely.)
> 
> The map has been there for a long time, the mapping widget is targeted at non ArduPilot users

Ok, I guess I never found it because I was too busy getting everything converted from QX7 to TX16S when I got the new radio a few months ago. However, I just checked the widget settings and also read the documentation and didn't manage to set up a map page. Is it a widget page, like the message history (2) and the new graphs (6)?

AlksSAV, 2021-07-06:
> > I use betian 800 GPS in Radiomaster tx16s.
> 
> 
> 
> A while ago I was looking for info on how to add a GPS module to TX16S and how to make use of it, but couldn't find anything. Do you have a link maybe?

I have connected GPS and Bluetooth to the equipment. I have a video on my channel(in Russian). The main thing is to choose the firmware internalgps, not internalbt.

https://youtu.be/0UulXKpc6Kg

yaapu, 2021-07-06:
@FasterRider 
[image]

this is from the wiki, assign a switch/knob to a channel (via the mixer page) then configure that channel as screen toggle in the Yaapu Config, once in the main screen by usiong the switch you'll be able to toggle between messages, hud and map

FasterRider, 2021-07-06:
> @FasterRider
> this is from the wiki, assign a switch/knob to a channel (via the mixer page) then configure that channel as screen toggle in the Yaapu Config, once in the main screen by usiong the switch you'll be able to toggle between messages, hud and map

Thanks a lot. But I can still use "page 5" as well? Because I'm always low on switches, and one is already needed for map zoom, so I'd prefer to use the Page button to switch between those views.

FasterRider, 2021-07-06:
> I have connected GPS and Bluetooth to the equipment. I have a video on my channel(in Russian). The main thing is to choose the firmware internalgps, not internalbt.
>
> https://youtu.be/0UulXKpc6Kg

Thanks, I'll check it. In fact I recently came across several AP-related videos in Russian that seemed to be very interesting, but unfortunately the automatically translated subtitles didn't make much sense.

yaapu, 2021-07-06:
> Thanks a lot. But I can still use "page 5" as well? Because I'm always low on switches, and one is already needed for map zoom, so I'd prefer to use the Page button to switch between those views.

sure, that will work

FasterRider, 2021-07-07:
> > Thanks a lot. But I can still use "page 5" as well? Because I'm always low on switches, and one is already needed for map zoom, so I'd prefer to use the Page button to switch between those views.
> 
> sure, that will work

Thanks, so to have an overview, the pages are:

1 - Main
2 - Message History
3 - ?
4 - ?
5 - Map
6 - Graph

And is there already a way to get the maps/tiles live via mobile network (around phone position until GPS is installed)? Because only now I read about the Google Maps option and downloading tiles via MP. I had used the older gmapcatcher method according to some YT video. So now I wonder if maybe I'm again behind recent developments, and maybe could skip the tiles downloading altogether.