yaapu/FrskyTelemetryScript #121: Truncated messages at Taranis Lite screen

RMFPV, 2021-08-24:
System messages from ardupilot at the second screen are truncated because they are longer than the screen.
Is it possible to move cropped part to the next line?

Thank you.

yaapu, 2022-10-17:
hi, check latest dev version (dev branch)

closed 2022-10-17 by yaapu

RMFPV, 2022-10-19:
Thanks a lot! Looks good!

Just noticed that sometimes the last string gets replaced with the following one.
[image]

And here is how it looked before replacement:
[image]

yaapu, 2022-10-20:
Mhmm, I can't replicate it, that could really be a telemetry issue dropping a piece of a message, I'm pretty sure the code is ok, I did extensive simulation on this feature which is simple but tricky :-)

RMFPV, 2022-10-20:
It happens only with exactly these two messages, all the others seems fine.