yaapu/FrskyTelemetryScript #128: Problem with Yaapu telemetry widget using CRSF and ArduCopter 4.1 RC2 and RC3

joe-stroup, 2021-09-28:
Hello from Atlanta -  I was about to post a problem with CRSF telemetry (Yaapu) using Copter 4.1 RC2, but noticed RC3 in the release notes.  Just installed it - no help.

Yaapu runs fine on my TX16s under FrSky and ArduCopter - so I'm confident I have it installed ok on my transmitter.  I'm using Yaapu 1.8.1.

Using a Crossfire diversity receiver (the large one) and CRSF for RCIN via the TELEM1 port, I get both the 23 normal and 10 additional "passthrough" sensors when I discover sensors.  

However quickly after the additional 10 passthrough sensors show up - the radio announces "telemetry lost" and they go from black to red on the transmitter display screen.  Yaapu reports "No Telemetry Data" on power up.

Thanks for any help - might not be a bug - but something I've done wrong.

yaapu, 2021-09-28:
You need to use the latest dev version for CRSF support https://github.com/yaapu/FrskyTelemetryScript/archive/refs/heads/dev.zip

closed 2021-09-28 by yaapu

jonas-koeritz, 2021-11-16:
I am running the latest dev of your script, Ardupilot 4.1.0 and ELRS 2.0RC2, it kind of works, but telemetry is really slow (text log messages are rare/non existent). I really like the direction that ELRS is heading and just removed a Tracer RX (1:1 swapped it for a Happymodel EP). Any hints on what else I might need to configure? Had it working on Tracer before.

yaapu, 2021-11-16:
which rates are you using in ELRS?
Without this PR https://github.com/ArduPilot/ardupilot/pull/19264 the only rates that actually work are 100:2 and 200:2, 200:4 anithing slower will not work.
With the PR 50:2 is usable

jonas-koeritz, 2021-11-16:
> Without this PR ArduPilot/ardupilot#19264

Where do I configure those? The ELRS Lua just offers values like "1:64"
(note the "wrong" order).

2021, 18:58:

yaapu, 2021-11-16:
100:2 ==> 100Hz with 1:2 ratio :-)

jonas-koeritz, 2021-11-17:
I just wasn't familiar with the terminology. I can't select 100 Hz, but 150Hz at 1:2 works like a charm. Thanks for your help! I now can finally ditch my tracer stuff! It's amazing that I am now able to run (almost) 100% Open Source Software and it's even better than before :)