yaapu/FrskyTelemetryScript #131: Sonar plotter

AlksSAV, 2021-10-30:
Tried out the latest version everything works fine. (1.9.5) Thank you very much!
#110  
If possible, add the function of shading the graph that is below the line, for a more realistic bottom.
Below are photos of examples.
[image]
[image]

yaapu, 2021-10-30:
Hi, it's possible but it will probably cause the radio to kill the script for excessive CPU usage, my script has a lot going on in the background, I will give it a try and see, can't say when though :-)

yaapu, 2021-10-30:
...and it wouldn't allow to plot 2 graphs

AlksSAV, 2021-11-13:
> Hi, it's possible but it will probably cause the radio to kill the script for excessive CPU usage, my script has a lot going on in the background, I will give it a try and see, can't say when though :-)

I understand you. Device stability is more important.

closed 2021-11-13 by AlksSAV