yaapu/FrskyTelemetryScript #133: Data under telemetry, but nothing visualilzed in yaapu

tkazik, 2021-11-28:
Hi!
First of all: very cool work, thx a lot!
I am trying to set up your telemetry scripts according to [this](https://youtu.be/x4ZIeLuQyXw?t=94) guide. The issue I am running into:

On the RC, I get plenty of telemetry data:
[image]

But nothing gets displayed in the widget:
[image]

Any hints for further troubleshooting? Thx!

My setup:

* Radiomaster TX16s with OpenTX 2.3.14
* Jumper R8
* Pixhawk 4 Mini with latest stable arducopter

yaapu, 2021-11-29:
Hi,
you need to set SERIALn_PROTOCOL = 10 for sport or SERIALn_PROTOCOL=23 for fport, I don't have that receiver so can't help 

From the screenshot it looks like you set SERIALn_PROTOCOL = 4

The official ardupilot docs are a better source and updated often
https://ardupilot.org/copter/docs/common-connecting-sport-fport.html#common-connecting-sport-fport

tkazik, 2021-11-29:
Thx a lot for your fast and on spot answer...solved it.

closed 2021-11-29 by tkazik