yaapu/FrskyTelemetryScript #136: Taranis on 128x64 : Msg length is limited to 16/23 char / Unable to follow AutoTune Messages

romarysonrier, 2021-12-26:
Hello,

First of all, being able to run the yaapu script V1.9.5 on a Jumper T-Lite Radio is a awesome !

When Using Ardupilot, the AutoTune messages are quite cyclics, so the similar message are aggregated and thus limited to 16 char... which makes them almost useless. Even 23 char is too small to be able to follow the Autotune messages.

My proposal is to add a config option name "shorter message prefix" , having the following consequences :
-  Disable Message aggregation ( to have a fixed max message length )
-  Disable "message number prefix" (not so usefull)
-  Shorter Severity prefix : Only the first letter is enought ( WARN gives "W", Alert => A , Crit =>and so on ..)

Thus we will increase a lot max message length from Ardupilot !

PS : it would be great to have scrolling character for long messages.

yaapu, 2021-12-31:
Hi, next release will have this feature

closed 2021-12-31 by yaapu

Johnex, 2022-06-06:
Is this available in a dev version to test? :)