yaapu/FrskyTelemetryScript #137: No Telemetry Window Blank on EdgeTX

davidbitton, 2022-01-07:
The red window that shows the "No Telemetry" shows up -- on my Radiomaster TX16S running EdgeTX--  as an empty black box. I installed the latest from the dev branch. Thanks.

edernucci, 2022-01-13:
Since you didn't mention your RX and are using `EdgeTX` and a very new radio, I suppose you're running `ExpressLRS`. If not, just ignore this message and please update your info with your receiver brand and model.

Press the `SYS` button and go to `Yaapu Config` under `TOOLS` menu. Scroll until find `enable CRSF support` and change it to `yes`. Reboot.

Find [more information here](https://github.com/yaapu/FrskyTelemetryScript/wiki/Passthrough-over-CRSF-and-ExpressLRS).

yaapu, 2022-01-13:
@davidbitton you need version 1.9.6 https://github.com/yaapu/FrskyTelemetryScript/archive/dev.zip, 

the wiki home has a note about it next to "EdgeTX support" ;-)

closed 2022-01-13 by yaapu

davidbitton, 2022-01-13:
@yaapu i'm using whatever is in github under the dev banch. Perhaps I should do another pull.

yaapu, 2022-01-13:
mhmm which EdgeTX version?

davidbitton, 2022-01-13:
@yaapu 2.5.0

yaapu, 2022-01-14:
that absolutely works, I have it on my radio. Please double check you're using widget version 1.9.6