yaapu/FrskyTelemetryScript #143: Horus 1.9.5: battery % by voltage issue

i3dm, 2022-02-22:
Horus V1.9.5:
using PX4 and one power module on Power 2 of PX4.
Yaapu config:
dual battery config - dual with alert on B2 (didnt find any other way to work with B2 only)
enable battery % by voltage: YES
Battey 2 count override: 3 cells

expected behaviour: battery % changes according to pack voltage displayed.
actual behaviour: battery shows 99% regardless of voltage (12.6V down to 10.0V)

yaapu, 2022-03-13:
Hi, please excuse the very late reply.
The battery % by voltage requires an extra lua file where you specify the discharge curve.
The script looks for a file named modelname_batt.lua in the cfg directory (same place as the script config file)

the content is:

-------------
```
local voltageDrop = 0.6

local useCellVoltage = false;

local dischargeCurve = { 
  {3.48*12, 0}, 
  {3.51*12, 10}, 
  {3.58*12, 20}, 
  {3.66*12, 30}, 
  {3.76*12, 40},
  {3.81*21, 50},
  {3.88*12, 60},
  {3.93*12, 70},
  {4.03*12, 80},
  {4.08*12, 90},
  {4.20*12, 99},
  }

return {voltageDrop=voltageDrop,useCellVoltage=useCellVoltage,dischargeCurve=dischargeCurve}

```
-------------

The script uses an 11 points discharge curve mapped into the array 
voltageDrop acounts for under load voltage sags, adjust per your needs
useCellVoltage specifies if the 11 values you enter in the array refer to cell voltage or battery voltage

i3dm, 2022-03-13:
> Hi, please excuse the very late reply. The battery % by voltage requires an extra lua file where you specify the discharge curve. The script looks for a file named modelname_batt.lua in the cfg [...]

thank you.
why are you not including this file by default?

yaapu, 2022-03-13:
good question, I guess I forgot to include it, but will asap

i3dm, 2022-03-15:
i added the attached lua (zipped here) file in the CFG folder.
configured bat % by voltage to YES
voltage of both the pack and the lowest cell is quite low.yet still showing constant 99%
any ideas?
[modelname_batt.zip](https://github.com/yaapu/FrskyTelemetryScript/files/8256267/modelname_batt.zip)
[image]
[image]

yaapu, 2022-03-16:
hi, did you rename the file based on your actual model name just like the .cfg file?

i3dm, 2022-03-16:
> hi, did you rename the file based on your actual model name just like the .cfg file?

i forgot, thank you. will test it soon and report.

i3dm, 2022-03-16:
Renamed the file, and now it shows a constant 00%.
here is the final renamed lua file:
[s550_batt.zip](https://github.com/yaapu/FrskyTelemetryScript/files/8261640/s550_batt.zip)

and a photo of CFG folder:
[image]

and the widget settings and result:
[image]
[image]

any ideas?

yaapu, 2022-03-16:
Your lua file is fine, seems like you spotted a bug in the 1.9.5 version, I tested in my dev branch and it works, here's the latest dev version for you to check
[SD.ZIP](https://github.com/yaapu/FrskyTelemetryScript/files/8261844/SD.ZIP)

i3dm, 2022-03-16:
> Your lua file is fine, seems like you spotted a bug in the 1.9.5 version, I tested in my dev branch and it works, here's the latest dev version for you to check [SD.ZIP](https://github.com/yaapu/FrskyTelemetryScript/files/8261844/SD.ZIP)

may i bother you to know which files do i need to replace to install the new version without messing up my existing configuration?
will just the WIDGET folder be enough?

yaapu, 2022-03-16:
everything but the CFG,

i3dm, 2022-03-16:
> everything but the CFG,

over written all files (but the CFG) and still im getting 00%

i3dm, 2022-03-19:
> > everything but the CFG,
> 
> over written all files (but the CFG) and still im getting 00%

any ideas?

i3dm, 2022-03-24:
> everything but the CFG,

Dear Yaapu.
any other ideas why its not working?
can you share your working configuration?

JEFlatt53, 2022-03-28:
Hello, I am having a similar issue in 1.9.6dev where the percentage displayed does not align with what I have in the cfg file for my model (using the lookup table you attached here). Per cell voltage is 4.11v but I am getting a 65% reading.

JEFlatt53, 2022-03-28:
> Hello, I am having a similar issue in 1.9.6dev where the percentage displayed does not align with what I have in the cfg file for my model (using the lookup table you attached here). Per cell voltage is 4.11v but I am getting a 65% reading.

It seems my issue was the voltage drop parameter. I think you can disregard this.

i3dm, 2022-04-13:
@yaapu  can you please advise about my issue?

Nerenye, 2022-05-13:
Is there a possibility to enable that on Ethos too? 
Since there is no CFG folder i?

yaapu, 2024-10-10:
@i3dm I'm sorry, I totally missed your reply