yaapu/FrskyTelemetryScript #145: Telemetry is loaded in the model setup but doesn't update and GPS stops working.

CamGenius, 2022-03-14:
I have my autopilot setup with fport protocol=23 and and am able to load the telemetry on to my TX16S with the 1.9.6 Yappu script but it doesn't show any updates to the telemetry or on the script. Also about 10 seconds after powering on the autopilot GPS sensor coordinates are lost and never recovered. Any help with this would be much appreciated
[image]
[image]

yaapu, 2022-03-16:
Hi, this has nothing to do with my script, it looks like an ardupilot telemetry issue.
Try switching serial port and double check your serial options, you do not mention your board model and how you wired it

Avalonnw, 2022-06-14:
Which receiver are you using?

yaapu, 2022-10-17:
I'm closing this, if you have not solved your issue please reopen it

closed 2022-10-17 by yaapu