yaapu/FrskyTelemetryScript #150: Cloning ELRS model makes telemetry nor work on clone

skorokithakis, 2022-04-20:
I'm running the widget on EdgeTX, I have an ELRS model running AP that's worked great for ages. I cloned that model to make some changes, everything works on the clone (I get telemetry sensors, etc), but the widget keeps saying NO TELEM.

The original model works fine still. I've removed the widget and added it again, set crossfire to on, but no dice.

skorokithakis, 2022-04-21:
I've managed to come up with steps to reproduce:

1. Clone a model
2. Use EdgeTX configurator to change the new model's name
3. Write it back
4. Yaapu telemetry stops working

urlu75, 2022-04-22:
I noticed the same behavior. To make it work again, at least in my case, I just need to set it up again. It seems that when the name changed, yaapu create a new blank cfg file with the new name. Yaapu config file does not follow model name change.

I also noticed with EDGE > 2.6 that ELRS lua script is stuck in loading phase.
When yaapu script is disable, it works straight.
When yaapu telem is enabled, to make the ELRS config lua script works, I need to make a change in the yaapu config (whatever is the change) and after saving new conf, ELRS script works just right after.

skorokithakis, 2022-04-22:
I *think* I got it to fail even when just changing an unrelated parameter in the configurator. If I rename the model from the radio, it works fine, but the configurator breaks it.

yaapu, 2022-05-03:
Hi, sorry for the delay, been really busy.
The config file is created only when using the config menu as you may have noticed. Config file name follows model name but does not "detect" name changes, so if you change the model after a config chnage the config file will still be named with the older model name.
At start the widget looks for the config file using the current model name so if you renamed your model it will not find it and if you use non default values it will not work (i.e. CRSF).
No other dependencies so not sure how changing other parameters might affect the widget

skorokithakis, 2022-05-03:
I don't know if this is strictly related to the name, because AFAIK it also happened when I made a mix change in the configurator (without changing the name). Should going into the config menu create a separate config file with the changed menu, perhaps?

yaapu, 2022-10-17:
@skorokithakis I'm closing this, it's been idle for a long time, I suppose you solved your issue

closed 2022-10-17 by yaapu