yaapu/FrskyTelemetryScript #151: Failsafe waring on Flight deck but no fail safes on ground station

lunerblade, 2022-04-25:
I am running PX4 the aircraft flys fine but after arming I get a failsafe waring they never goes away. The ground station shows no issues no failsafes trigger in flight. Since all it says is failsafe im not sure what to look at or change to make this stop.

yaapu, 2022-05-03:
this is not related to my script sorry. Since you use PX4 you must be using some kind of converter between mavlink and frsky so please check that part of your setup.

closed 2022-05-03 by yaapu

drnry, 2022-05-14:
> I am running PX4 the aircraft flys fine but after arming I get a failsafe waring they never goes away. The ground station shows no issues no failsafes trigger in flight. Since all it says is failsafe im not sure what to look at or change to make this stop.

Hi, did you figure out what's causing it? I have the same issue and it came up after upgrading yaaou script to the latest version - nothing else changed.

lunerblade, 2022-05-14:
Sadly I did not.

yaapu, 2022-09-07:
Hi, after more checks latest yaapu version added generic failsafe at offset 12 on frame 0x5001, I suspect that MavToPT uses that bit for another purpose

yaapu, 2022-09-07:
Checked latest MavToPT source and that bit is correctely handled so you should upgrade to the latest MavToPT version

https://github.com/zs6buj/MavlinkToPassthru/blob/fa32fd7abc829c063a24f2cacf6cb1fc4d70bd70/Source/Stable/MavToPass/FrSky_Ports.h#L2154