yaapu/FrskyTelemetryScript #155: continuous restarting/initializing

david9981, 2022-05-25:
I am using latest version of Ardupilot/EdgeTx/Elrs and Yaapu widget Dev1.9.6. Radio is TX16s MKII.
Since no compass is used, it is normal and expected EKF logic that I constantly get "Bad AHRS" until after GPS origin is set.  Even then, I would still constantly get "Bad AHRS" message until yaw alignment is completed, which happens moments after airborne.

I tried to configure the alert message to appear every 10 sec (default) but it still continue to pop up fast.  The biggest issue is after every alert message, the widget would "restart" itself, initializing and loading layout, which makes it almost unusable on ground. Please assist. video attached.

[https://youtube.com/shorts/BB2MrHJUxGc?feature=share](url)

ardisd, 2022-06-03:
I am having the same issue with a radiomaster TS16 MKII running edgeTX version 2.7.1 Yaapu widget Dev1.9.6

yaapu, 2022-06-16:
Hi, edgetx changed the way global variables are handled from lua, please check the workaround here
https://www.rcgroups.com/forums/showpost.php?p=49299729&postcount=852

closed 2022-06-16 by yaapu