yaapu/FrskyTelemetryScript #158: Add support for Ethos on Horus X10S

greggibbons, 2022-06-11:
I have your Telemetry script running on my X20 with ETHOS and love it.
But I have two nice X10s I would like to install it on, will you be porting it to the X10s?

yaapu, 2022-06-16:
Hi, I ported the X20 script to full screen first and next is the X10 implementation

[image]

time on my side though is really limited lately

yaapu, 2022-10-06:
porting done, check the repo

closed 2022-10-06 by yaapu

greggibbons, 2022-10-07:
I imported your new port to both my X20 and my X10s, both radio Telemetry screens show about 1/4th of the full screen, in other words 1/4th of the full Yaapu Telemetry image takes up the whole radio screen, I could not find a setting to fix it, what should I do?
[image]
[image]

yaapu, 2022-10-07:
wrong layout, you need to select "Full Screen", check the new wiki section https://github.com/yaapu/FrskyTelemetryScript/wiki/Installation-on-Ethos

greggibbons, 2022-10-08:
Thank you, that worked.