yaapu/FrskyTelemetryScript #163: Can't have connection with my ELRS 2.4 Namimno on my Horus X10 with 1.9.6 Frskytelemetriescript

captflame2121, 2022-07-20:
Hi, I'm in ELRS 3.0 on Namimno ELRS 2.4 .
My Horus X10 is in elrs 2.7.1
My FC is with Bétaflight  4.3  

For use Frsky telemetrie 1.9.6 you can use BF or you must in ardupilot ?

If you can use BF :
I have put all on my SD  .
I can play Yappu  BUT my horus can't connect on my Elrs Namimno .

I have create 3 files:
mrcroc.cfg
mrcroc_right_sensors.lua
mrcroc_sensors.lua

Can you help please, I 'm sorry my English is bad .   Best regards

yaapu, 2022-08-16:
Hi, my telemetry scripts and widgets require ArduPilot, they do not work with betaflight

closed 2022-08-16 by yaapu

captflame2121, 2022-08-16:
Hi Alessandro ,
Thanks for your answer  but I had find .  Yes with Ardupilot.

BEST REGARDS

FLORENT

yOUR WORK IS FANTASTIC , THANKS

Envoyé à partir de Courrier<https://go.microsoft.com/fwlink/?LinkId=550986> pour Windows

Envoyé le :mardi 16 août 2022 23:22
À : ***@***.***>
Cc : ***@***.***>; ***@***.***>