yaapu/FrskyTelemetryScript #164: Yaapu telemetry "initializing/loading layout" constant error

DarioHxC, 2022-07-22:
Every time I flip the arm or mode switch, I get this error "initializing/ loading layout" as you can see in the video. Any help would be really appreciated. 

https://youtu.be/2YAY7X4GNIE

Johnex, 2022-08-11:
I have the same on EdgeTX 2.7.1. The Lua script works fine but every time a message arrives, it resets like @DarioHxC had.

@yaapu Is this a known issue so far?

DarioHxC, 2022-08-12:
@Johnex havent got a hold from nobody. I think is not a common error since I haven't found anything online. I even tried in the EdgeTx group on FB and nobody said anything so far...

txmrb, 2022-08-14:
same issue for me, Edgetx 2.7.1, ExpressLRS 2.5.1 and Yaapu Version 1.9.6 (current dev version). Since I am using the dev tree for ExpressLRS support I didnt report the issue. What yaapu Versions are you running?

Johnex, 2022-08-14:
EdgeTX v2.7.1, Crossfire v6.19, Arducopter v4.2.2, Yaapu 1.9.6 dev.

DarioHxC, 2022-08-14:
All the latest and greatest.

txmrb, 2022-08-14:
Do you have the same issue on the stable Yaapu Version? My R-XSR died and I have only eLRS telemetry receivers so can't downgrade to 1.9.5. If the same issue appears on 1.9.5 it would point to EdgeTX rather than Yaapu

DarioHxC, 2022-08-14:
I have a QX7 on OpenTx running old versions (Yaapu and Otx) that works just fine with CRSF and an old RXSR. I don't know who to blame...

txmrb, 2022-08-14:
I mean @Johnex since we both have the same EdgeTX version, same Yaapu Version and same issue. So if he can reproduce the issue on Yaapu stable we'd know its something with EdgeTX and not the script itself. Just to get some more details for the developers.

DarioHxC, 2022-08-14:
Send it!

txmrb, 2022-08-14:
@DarioHxC let's keep the Comments on Topic, doesn't help if the maintainers have to scroll trough a lot of offtopic talk

DarioHxC, 2022-08-14:
You'd better delete this last comment then...

Johnex, 2022-08-14:
I will test in a few days, but i remember that the stable version doesn't have the CRSF variable sized packets, so it won't work properly (or at all) with the 6.19 crossfire v3 protocol.

Also, there was an issue with Yaapu and EdgeTX because of a how Yaapu was using the GVars. Seems like it's the issue we are seeing, but the EdgeTX guys say it's fixed/working? 

https://github.com/EdgeTX/edgetx/issues/2021

It's hard to understand if they are labeling "working with restarts" as working?

Johnex, 2022-08-14:
Ahh, there is a fix for the issue here:

https://www.rcgroups.com/forums/showpost.php?p=49299729&postcount=852

DarioHxC, 2022-08-15:
@Johnex Thanks for the investigation work. Have you tried it yet? Did it work?

DarioHxC, 2022-08-15:
@Johnex it works!

yaapu, 2022-11-26:
solved by Horus Version 2.0.0 beta2, no more need for GVars

closed 2022-11-26 by yaapu

DarioHxC, 2022-11-26:
Great work! Thanks for taking the time. So should I install that version?

yaapu, 2022-11-27:
@DarioHxC give it a try, should be pretty stable, I did many hours of simulated flights

DarioHxC, 2022-11-27:
Grazie mille, Alessandro!