yaapu/FrskyTelemetryScript #171: How to set FrSky Taranis transmitter to get telemetry on screen ?

Blackdevil1691, 2022-09-07:
Hello there, 
I want to get telemetry data on my Taranis Access transmitter. I have an X8R receiver and Pixhawk 2.4.8 with a built-in F4 processor. After my research, I understand that we need to flash or run the LUA script in the Taranis transmitter. 
Can anybody tell me how to set up this and how to get telemetry data on my transmitter screen? 

**Regards, 
Blackdevil**

yak-54, 2022-09-08:
So what is the Issue with yaapu telemetry ?
you have open an new issue so what's the issue your having ?

Blackdevil1691, 2022-09-08:
Hello Colin, 
I want to know how to set up this Yaapu telemetry cable to get telemetry data on the Taranis screen, I am completely blank and didn't find anything fruitful.
Could you please tell me how to get telemetry data like posted on this repository?

yak-54, 2022-09-08:
> Hello Colin, I want to know how to set up this Yaapu telemetry cable to get telemetry data on the Taranis screen, I am completely blank and didn't find anything fruitful. Could you please tell me how to get telemetry data like posted on this repository?

Have read here and ask your questions 

https://discuss.ardupilot.org/t/an-open-source-frsky-telemetry-script-for-the-horus-x10-x12-and-taranis-x9d-x9e-and-qx7-radios/26443

Blackdevil1691, 2022-09-09:
Yet, I understand that I have to flash/run the LUA script in Taranis. there is no procedure which I can follow to set this to get telemetry data

Blackdevil1691, 2022-09-15:
Hello, please revert. I am unable to find any support online.

yaapu, 2022-09-23:
Hi read here https://github.com/yaapu/FrskyTelemetryScript/wiki/Installation-on-Taranis-radios

closed 2022-09-23 by yaapu