yaapu/FrskyTelemetryScript #172: Edge TX 2.7.1 with widget Edget TX 1.9.6: change page of widget doesn't work

macfly1202, 2022-09-11:
Hi, Going from Horus version on OPEN TX to Edge TX. I updated the Yaapu script and it seems that page of widget can't be change anymore with switch on radio TX16s. It was working like a breeze on OpenTX. Does this feature is broken on Edget TX ?

yaapu, 2022-10-17:
hi, could you please test version 2.0.0 https://github.com/yaapu/FrskyTelemetryScript/archive/refs/heads/dev.zip

macfly1202, 2022-10-17:
Hi, will test this tomorrow evening and report feed back. thanks !

macfly1202, 2022-10-18:
Hi, quick feedback. From what I tested but because it's the night/windy, I didn't gone outside to test the maps feature with GPS signal (that was working perfectly on 1.9.6). Everything seems smooth, change screen and zoom level work as expected.
FPS is 6.8 on avionics screen and ~12 FPS on mavlink log screen. Another ciffer below FPS is 779049.
Thanks again for imprvements. I will test GPS and maps this WE if weather ok.

macfly1202, 2022-10-19:
Some screenshots 

[image]

[image]

[image]

[image]

yaapu, 2022-10-20:
Looking good, thanks!

closed 2022-10-20 by yaapu