yaapu/FrskyTelemetryScript #174: Yaapu 1.9.6 - Incorrect flight modes

matador2437, 2022-10-01:
Gotta be something simple (I hope), but Yaapu V1.9.6 is not reading the flights correctly. This is what I'm running:
Pixhawk 5X, TX16S (6pos. switch), Edge TX 2.7.1, ExpressLRS V3.0, Yaapu 1.9.6, Mission Planner 1.3.77, ArduCopter V4.2.3

Flight Mode           Yaapu Display          Yaapu Track Played
Stabilize                 AltCtl                       Altitude Control Mode
AltHold                  Acro                        Acro Flight Mode
Loiter                     RAttitude                Rate Attitude Mode
PosHld                   Land                        Land Flight Mode
Land                      (Blank)		     (Nothing played)
RTL                        Simple                     Simple Mode

Does anyone have a clue what could be the problem. Would much appreciate your help. Thanks.

lanyiuav, 2022-10-06:
i have no ideal ，as you can see it stop at1.9.6dev

IcemanDZ, 2022-11-11:
mine also have same issue, edgetx 2.8 RC2, YAPPU 1.9.6 Dev, arduplane 4.3.1 TBS crossfire.

Damian650, 2024-04-25:
Hello, first of all I would like to apologize for my English, I know the question is old, but I am having the same problem and I don't want to generate another post.
When I change the modes from my transmitter, what the telemetry shows does not match what is actually selected in the mission planner.

I am using version 2.0.0 on the transmitter, and the flight controller is a pixhawk 2.4.8 with arducopter version 4.5.1, the telemetry is Sport from Frsky, the receiver is the X8R series.

Thank you very much, if this happened to someone and they were able to solve it, I appreciate your help.

Damian650, 2024-04-26:
Hello again, I found the solution to the problem, perhaps this solution was already given somewhere else, although I managed to find it.

Without further ado, the solution is the following:

In my case, I use pixhawk, you have to go to the SD card, then Widgets, yaapu and lib. Inside there you will find a file called copter_px4.lua, inside there you have to modify part of the code, specifically what is under plane flight modes, there you will have to modify the names of the modes that you have programmed, for my case:
flightModes[1] = "Stabilize"
flightModes[2] = "AltHold"
flightModes[17] = "PosHold"
flightModes[6] = "Loiter"
flightModes[5] = "RTL"
flightModes[4] = "Auto"

If they are not ordered correctly in this way, they can change what the telemetry announces, so what it should really announce.

I hope it helps, needless to say, make a backup copy of the file, I don't know if it is the correct way to do it but it worked for me.

I am not responsible for any damage that may be caused to the hardware, to the individual or to third parties.

Hugs

yaapu, 2024-10-10:
> Hello again, I found the solution to the problem, perhaps this solution was already given somewhere else, although I managed to find it.
>
> Without further ado, the solution is the following:
>
> In my [...]

I closed it but your solution should not be necessary, the script/widget detects the vehicle type and the flightmode automatically.
If you had to edit the copter_px4.lua file it means that you accidentally enabled the px4 flightmode options inthe menu which is NOT for ardupilot but for PX4 firmware.

closed 2024-10-10 by yaapu