yaapu/FrskyTelemetryScript #175: Suppress late coming statustext beeps

olliw42, 2022-10-07:
with a mavlink to passthrough converter on the tx side it can happen that the statustext messages are much faster displayed than the notification beeps can follow (since statustext messages can come in quickly with mavlink but each beep requires a certain to play). Especially at startup or other periods of many statustext messages in a row this can add up to quite a number of seconds.

I find this very confusing, since it gives the impression that statustext messages are still coming in, but "somehow" are not displayed, i.e., missed. I think there should be no beeps anymore then there are no messages anymore.

I guess a simple workaround could be to simply clear all pending beeps when a statustext is displayed.

yaapu, 2022-10-07:
Hi @olliw I don't know of a way to clear the playTone() queue from lua, I used to have some code in place to limit the playTone rate up to a max so to not kill the radio memory by having a too large queue, so basically queue playTone uo to one every let's say 500ms and not more, I removed it for I needed the memory for other stuff on X9D radios :-)

olliw42, 2022-10-07:
ah, I see
I guess the time guard would then another simple option
surprising it takes so much resources, would be simply 
if (cur_time - last_beep_time > 50) then beep; last_beep_time = cur_time; end
wouldn't it
maybe one could do it for the horus, if not also for the x9d

btw, I am olliw42 :) there is also a OlliW github user, but that's not me ... he might be a bit annoyed now LOL

yaapu, 2022-10-08:
Yes it's a one line hack but on X9D I'm down at the "byte" level for saving memory, so I removed a lot of stuff, anyway you might try it yourself and  if it works with your use case I can simply add it back in, at least on color radios and Ethos.

yaapu, 2022-10-17:
I added a 2 msg/sec limit and unless I made a mess should solve your issue :-)
I updated the dev branch

closed 2022-10-17 by yaapu