yaapu/FrskyTelemetryScript #178: Is it possible to send MAVlink telemetry to a serial port in OpenTX?

arikrupnik, 2022-11-26:
I understand that LUA has access to serial port(s) in OpenTX/EdgeTX. Is it possible to forward MAVLink that comes in through FrSky telemetry to a serial port on Taranis or TX16S, so that the user could then forward that to a laptop? The alternative is to have a separate telemetry radio onboard (e.g., SiK) which feels silly since telemetry is already streaming down over SPort.

yaapu, 2022-11-26:
Hi, there's no mavlink coming in so no way to forward it.
This widget uses custom frsky telemetry as documented here https://docs.google.com/spreadsheets/d/1oFv5zSl10wyR0LOcCChDdy9peIRNkbaEBrsHDQ0ZdmE/edit#gid=0

closed 2022-11-26 by yaapu

arikrupnik, 2022-11-26:
Thank you for explaining this, Alex. I had the wrong idea about how this worked!

AlksSAV, 2022-11-27:
> I understand that LUA has access to serial port(s) in OpenTX/EdgeTX. Is it possible to forward MAVLink that comes in through FrSky telemetry to a serial port on Taranis or TX16S, so that the user [...]

To do this, simply connect bluetooth to the radio, and select telemetry transmission in the settings. The script is not needed here.

yaapu, 2022-11-27:
> To do this, simply connect bluetooth to the radio, and select telemetry transmission in the settings. The script is not needed here.

...what you get is a frsky telemetry relay which will not work if using my script in parallel because my script uses custom telemetry frames, you would need to enable SERIALn_PROTOCOL 4 (unless out there there's a telemetry app that understands passthrough telemetry I'm not aware of)

arikrupnik, 2022-11-27:
> ...what you get is a frsky telemetry relay which will not work if using my script in parallel because my script uses custom telemetry frames, you would need to enable SERIALn_PROTOCOL 4 (unless out [...]

I could write a decoder for my Python, but I don't think this approach is going to work for my application, which includes a PTZ camera. I understand that GPS effectively updates at 0.5 Hz (1 Hz but alternating lat/lon). This is too slow to drive a camera. It was a neat idea, but I don't think it's going to work. I'm back to two separate radios.