yaapu/FrskyTelemetryScript #179: Timer 3 mode OFF after restart

SkyTakes, 2022-11-28:
Radiomaster TX16S + Edgetx 2.8 + Yaapu Telemetry Widget 2.0.0 beta2 (for ardupilot)

I´ve set 3 timers on the radio with some Logic Switches and I noticed that everytime I turned the Radio off and then on the Timer 3 came up turned OFF (Mode changed from ON to OFF)

Then I started playing to see what was causing it and when I removed the YAAPU widged this behavior stopped.

So I created a blank plane model, turned the 3 timers ON and added the YAAPU Widget and bam!! Same bug. As soon as I power up the radio, timer 3 is OFF, the other 2 timers keep the config (Mode ON)

I just installed the last night widget and the bug is still there.

yaapu, 2022-11-29:
Timer 3 is used by the script to track, well, flight time :-) you can't use it for your own purposes

SkyTakes, 2022-11-29:
Ohhh I see... that makes sense :-)
Thanks for the reply and thank you for the amazing widget!!!

closed 2022-11-29 by yaapu