yaapu/FrskyTelemetryScript #181: Armed/disarmed information overwriten by battery failsafe

booo, 2022-11-30:
If the plane is in battery fail safe mode or battery fail safe was triggered this information overwrites the information about armed/disarmed state of the plane. E.g. you can't easily tell if you disarmed the plane after landing if the battery fail safe was triggered because the display shows "Battery fail safe" or something similar all the time.

booo, 2023-03-19:
We observed this on the color display of a radiomaster tx16s too.

This is super annoying and endangers our volunteer crew that operates our search and rescue UAVs. Due to this bug the pilots have a hard time to tell whether the UAV is disarmed or not.

Any hints how to fix this? We would love to contribute a solution.

yaapu, 2024-10-10:
this was fixed a while ago, thanks for pointing it out!

closed 2024-10-10 by yaapu

booo, 2024-10-11:
Thanks for your work! Much appreciated. We will try to update our version of the script and let you know if something else pops up.

yaapu, 2024-10-11:
...hold on, I just verified I did not publish the fix for the X9D, damn, only for color lcd radios