yaapu/FrskyTelemetryScript #182: Voltage display gets stuck on one value

booo, 2022-12-01:
During extensive test flights with our system we encountered from time to time problems with the battery voltage display. The voltage got stuck at some point during the flight and never changed over the course of the flight.

QGC/GCS still display the proper value of the battery.

Observed on a taranis X9D.

yaapu, 2022-12-01:
Hi, what's your setup? I suspect you loose the telemetry link

booo, 2022-12-01:
We have a Taranis X9D and a r-xsr receiver. We use a recent ArduPilot version.

Totally possible that we loose telemetry link from time to time. Other telemetry data returns at some point and gets displayed properly.

We didn't have time to collect more information about the problem while doing the flights. I will take a closer look the next time we do flights.

yaapu, 2023-11-08:
stale for too long, perhaps reopen it if required

closed 2023-11-08 by yaapu