yaapu/FrskyTelemetryScript #183: Total distance calculation based on actual UAV position

booo, 2022-12-01:
Is it possible to calculate the total distance traveled based on the actual UAV positions? Do we receive the UAV position via the telemetry data? Or do we need an additional script or something on the FC as mentioned [here](https://github.com/yaapu/FrskyTelemetryScript/issues/96).

The [current calculation](https://github.com/yaapu/FrskyTelemetryScript/blob/dev/TARANIS/X9D/SRC/yaapu9.lua#L1309) differs from the calculations done by e.g. QGC quite a lot for flights longer than e.g. 50 kilometers.

yaapu, 2022-12-01:
Yeah, I agree, the calculation is pretty basic :-) I can switch method and use sampled gps distance, I can sure improve it, nobody actually bothered to ask for a better implementation, would you be willing to test it?

yaapu, 2022-12-01:
Switching to haversine I've got an error of 200m on a 75Km flight, should be good enough (reference is ArduPilot OSD which is updated at 10Hz)

booo, 2022-12-01:
That sounds great. We can test the implementation on one of our next test flights for our [project](https://www.hs-augsburg.de/searchwing/).

Thanks for taking the time to implement a better solution!

yaapu, 2022-12-09:
hi, please try latest master

booo, 2023-01-18:
Sorry, this slipped through. We will do tests in the next weeks. Thanks for your work!

befut8, 2024-02-18:
did you had a chance to test ?