yaapu/FrskyTelemetryScript #185: Script gets stuck when switching models

Allister779, 2022-12-03:
When I change from one model to the next the script will play the "Yaapu Telemetry Ready" .wav and then the screen will stop on the "Initializing" message.  When I power cycle the radio the script will start without issue.  I've also tried using all the reset telemetry options.  It doesn't seem to matter if there is a drone connected or now.

Edge TX 2.8.0.  Radiomaster TX16S.  Yaapu 2.0.0 Beta2.  Radio links have been FrSky D16, FrSky X2 D16, and CRSF.  All the drones tested so far have been on ArduCopter, but I can test a few planes as well if by some chance it matters.

Thanks!

yaapu, 2022-12-06:
Hi, it should be fixed in latest master, download the zip, not the release

closed 2022-12-07 by Allister779

yaapu, 2022-12-07:
great, thanks for your feedback