yaapu/FrskyTelemetryScript #187: Another no Telemetry Issue

PeterQFR, 2023-01-09:
I've been following the multitude of guides, but it doesn't seem that telemetry is being passed to my RC controller

Hardware:
Holybro Kakute H7-bdsh (fits v2)
FRSky XSR reciever
Receiver Sport connected to T3
Reciever Sbus connected to R6

RC Hardware Taranis X9d+ with opentxt v2.3.15-otx

SERIAL3_PROTOCOL = 10
SERIAL3_OPTIONS = 7

SERIAL6_PROTOCOL = 23
SERIAL6 _OPTIONS = 3

Arducopter v4.3.0-dev

Yappu v1.9.5

When I run the Yappu debug script it shows zero telemetry being recieved. Are there any other configurations required? I notice there is a FRSKY_UPLINK_ID.. does that need to be set to anything? currently it is 7.

PeterQFR, 2023-01-09:
@yaapu I upgraded to Yappu v2.0.0-dev script as I realised the 1.9.5 was a bit old.

I do have the sqt5font enabled, will that prevent telemetry from working? The Lua script runs as far as I can tell. Also the version of openTX I am using doesn't have Multimode options, I am also not using the EU build option (and neither are my Radios configured for EU standards)

yaapu, 2023-01-11:
you need to get the telemetry working first.
Can you discover the GPS sensor under OpenTX, does RC work at all? If not your wiring must be double checked.
As I said this has probably nothing to do with setting up the telemetry script.
sport telemetry can be tricky unfortunately

PeterQFR, 2023-01-12:
I cannot get any additional sensors under OpenTX so I don't think telemetry is coming through at all. Do the SR3_* parameters control telemetry for passthrough? They are currently all at 0Hz.

RC works, But I run SBUS from the RXSR to Telem6 RX and configure serial6 as RCIN. I wasn't able to get Ardupilot to recognise RC as being connected when Seria6 was configured as FRSKY Passthrough Telemetry.
Thats why I run SPORT through Serial3 as configured above. (Serial 4 is GPS for me). 

I thought as one point that there might have been a configuration either in Taranis or the FrSky Radios for no Telemetry, have you come across it? I may also be confusing with something else, its only a vague recollection.

PeterQFR, 2023-04-21:
@yaapu Found the fix, simple if you remember it. I only found it as I had to rebind a controller. But during the binding procedure on OpenTX you get the option to do channels 1-8 with Telemetry or 1-8 with no telemetry. I guess I had previously chosen no Telemetry.

Selecting the option with telemetry, it immediately started working.

closed 2023-04-21 by PeterQFR