yaapu/FrskyTelemetryScript #189: Yappu9 stopped working since updating to OpenTx 2.3.15

saintofinternet, 2023-01-17:
Hi,

I recently updated my Taranis X9D+ to OpenTx 2.3.15 via the OpenTX Companion.

While I could select LUA in the Companion there was no LUAC option. So I proceeded with just LUA selected and downloaded the firmware and upgraded it. 

No my Yappu9 script has stopped working and keeps showing "No Telemetry" 

Tried everything and even tried changing the cable but still no luck. 

What could I be missing?

Image of the OpenTX Companion below with the options selected.

[image]

saintofinternet, 2023-01-20:
Hi, 
Thank you for the reply.

"Enable CRSF" meaning Crossfire?? As I don't have any Crossfire modules. I am using the Taranix X9D+ with it's internal module. 

I used the script from the https://github.com/yaapu/FrskyTelemetryScript page under the "Download Section"

I will try the Model Delete option as you have suggested.

yaapu, 2023-01-20:
download the latest using the zip option, not the releases option 
https://github.com/yaapu/FrskyTelemetryScript/archive/refs/heads/master.zip

closed 2024-10-10 by yaapu