yaapu/FrskyTelemetryScript #192: hello

KarDanenko, 2023-02-01:
hello, I really like your script, tell me how can I add 2 language packs there myself

yaapu, 2023-02-01:
which radio?

KarDanenko, 2023-02-01:
radiomaster tx16

yaapu, 2023-02-01:
ok, latest version (2.0.0 dev) allows to override 4 sound categories
- base sound files
- px4 sound files
- status text hash sound files
- aerobatics tricks sound files

Translate the 4 english ".psv" [template files](https://github.com/yaapu/FrskyTelemetryScript/tree/master/OPENTX_EDGETX/COLOR%20LCD/HORUS/SD/WIDGETS/yaapu/sounds) and use [TTSAutomate](https://ttsautomate.com/) to generate the new sound files

Create a new directory for each new language under /WIDGEST/yaapu/sounds, as an example for polish create /WIDGET/yaapu/sounds/pl

Add each new language as a menu option under
- /SCRIPTS/TOOLS/Yaapu Config.lua
- /WIDGETS/yaapu/menu.lua

change this line in both files

OLD: {"voice language:", "L1", 1, { "english", "italian", "french", "german" } , {"en","it","fr","de"} },
NEW: {"voice language:", "L1", 1, { "english", "italian", "french", "german", "polish" } , {"en","it","fr","de","pl"} },

KarDanenko, 2023-02-01:
sorry ran into a problem can't install version (2.0.0 dev) was using Horus Widget version 1.9.5

KarDanenko, 2023-02-01:
[image]
[image]

KarDanenko, 2023-02-01:
[image]

KarDanenko, 2023-02-01:
[image]
[image]

yaapu, 2023-02-01:
Are you using ArduPilot?

KarDanenko, 2023-02-01:
yes version 4.3.1

yaapu, 2023-02-01:
Curr, Capa and _Bat sensors where do they come from, which RC protocol and how do you wire it?

KarDanenko, 2023-02-02:
TBS

yaapu, 2023-02-02:
Set rc_options = current value + 256
Enable crsf support in the widget config.
You NEED widget version 2 for CRSF support.
Use a serial port that has DMA for both tx and rx

closed 2023-11-08 by yaapu