yaapu/FrskyTelemetryScript #198: Syntax error: success

Xaurin, 2023-03-20:
Hi,

Using: Latest BW 128x64 release.
When trying to save settings on Radiomaster BOXER showing syntax error: modelname.cfg success.

Can't change to CRSF protocol.
LUA debug script for CRSF ver. 1.9.5 showing no data.

aviphysics, 2023-05-04:
any news on this issue?

aviphysics, 2023-05-04:
Figured it out. You need to copy over the common files for your setup.

klehnst, 2023-07-25:
Can you explain in a little more detail how you solved the problem?

aviphysics, 2023-08-03:
> Can you explain in a little more detail how you solved the problem?
> 
> I have the same problem and did not understand your solution.

In the folder structure there are folders for specific radios, and also a folder with files common to all radios. You need both folders.

closed 2024-10-10 by yaapu