yaapu/FrskyTelemetryScript #201: airspeed telemetry item "ASpd" not working in EdgeTx

BatesKG, 2023-04-06:
Over the winter I updated my TX16S to EdgeTX, and the inserted ASpd variable no longer works. Looking at the "main.lua" script, I see you are now inserting the ASpd item if "airspeed" is enabled.  You had advised me how to add the ASpd telemetry item in OpenTx about a year and a half ago.

The first issue is that it is not detecting airspeed as being enabled, so it doesn't even add the ASpd telemetry item at all. If I comment out the check to see if airspeed is enabled, the variable then appears on the Tx, but it always has a value of 0.

The airspeed is displaying fine on the tx widget, so the value is getting to the tx. Just not to the telemetry value. I saw in the main.lua that in the "VFR" section of the code, you are using different memory locations to set the airspeed variable value? 
IE:

in the telemetry section:
-- once detected it's sticky
if bit32.extract(value,28,1) == 1 then
telemetry.airspeed = bit32.extract(value,10,7) * (10^bit32.extract(value,9,1)) -- dm/s
else
telemetry.hSpeed = bit32.extract(value,10,7) * (10^bit32.extract(value,9,1)) -- dm/s
end
if status.airspeedEnabled == 0 then
status.airspeedEnabled = bit32.extract(value,28,1)
end

in the VFR section:
elseif appId == 0x50F2 then -- VFR
telemetry.airspeed = bit32.extract(value,1,7) * (10^bit32.extract(value,0,1)) -- dm/s
telemetry.throttle = bit32.extract(value,8,7) -- unsigned throttle
telemetry.baroAlt = bit32.extract(value,17,10) * (10^bit32.extract(value,15,2)) * 0.1 * (bit32.extract(value,27,1) == 1 and -1 or 1)
status.airspeedEnabled = 1
end

If I set all instances to "bit32.extract(value,1,7)", then the airspeed in the Tx widget stops working as well.   So "bit32.extract(value,10,7)" appears to be the correct value but setting all instances to that value doesn't fix the ASpd telemetry value.  

Thanks for any assistance you can provide on getting the ASpd tx variable working in EdgeTx.

yaapu, 2023-07-20:
please forgive me I missed this one.
OpenTX to EdgeTX should be no issue, we need to look elsewhere. 
Did you update ardupilot and/or the widget?

BatesKG, 2023-07-20:
I updated the widget to the latest version when I moved the TX16S from OpenTx to EdgeTx as I replaced the SD card contents for the new system.   
The airspeed value is getting to the transmitter and is displaying in the widget, it's just not setting the telemetry item "ASpd" to the current value so that I can reference it with a Special Function for sounding a stall warning, or for turning off my reverse thrust when the speed gets below a certain threshold.   The ASpd item just sits at 0.
Thanks for looking at this!

Kevin

BatesKG, 2023-07-20:
I'm going to try making the same change I did in OpenTx to get it working.   I'll try it later tonight and let you know if I had any luck.   Thanks again.

BatesKG, 2023-07-20:
No.   That didn't work.    
Just fyi, I'm running the March 6th version of the code.   The current code creates the telemetry item, but it just sits at 0 knots......even if I hold my finger on the pitot tube to increase the value, go back to the widget, and then back to the telemetry page.   ASpd is 0 even though the widget screen was showing 30 knots.
Thanks,
Kevin

yaapu, 2023-07-21:
Hi, some pointer for further checks
Did you read the docs https://github.com/yaapu/FrskyTelemetryScript/wiki/airspeed-vs-groundspeed?
Did you enable the separate airspeed telemetry by setting FRSKY_OPTIONS bit 0 to 1 (it's a bit mask)
If so, do you get the dual horizontal speed on the left side of the hud?
[image]
upper is airspeed lower (with the small G) is ground/gps speed

BatesKG, 2023-07-21:
Thank you!!!     Setting the FRSKY_OPTIONS bit to 1 did it.   I haven't been using FRSky for quite some time now. ( using CRSF)  So, I didn't think I still needed the FRSky options turned on.   However, that now makes sense in regards to the 'airspeed enabled' setting you are pulling in your script.  ::)    Even if you are using CRSF or some other protocol, the FRSky bit still needs to be set to 1 if you want the airspeed variable to work.   The thing that confused me was that the airspeed was showing in the widget.  After setting that option, I now see both air and ground speed in the widget. 
Thanks again!    
Unless you want to add a final comment to this issue, I'll close it as resolved.    Have a great day,

Kevin

closed 2023-07-24 by BatesKG

yaapu, 2023-07-25:
great news, thanks for your feedback