yaapu/FrskyTelemetryScript #202: ETHOS script not showing GPS data despite it being in telemetry

jaroszmm, 2023-04-10:
I just started with Yaapu on ETHOS, using it on OpenTX with no bigger issues (apart from a lot of lag on AP/CRSF combo).
But on ETHOS I noticed something interesting.

The GPS Coordinates data is not displayed on the screen. On my first flight I noticed there was no data at all. On my second flight the data was only zeroes all the time.  When I used "reset widget" option the data was gone completely again.

However when I set up 2nd screen in ETHOS and used one widget to display telemetry GPS data, the data was present there, so I'm pretty sure the GPS data is being transmitted to Yaapu but not decoded? 

[image]

yaapu, 2023-07-06:
Hi, I'm sorry but have no way to reproduce your issue, I do not own the radio and on X20, X12 and X10 works fine, it also works on the X18 simulator, so really no clues here

jaroszmm, 2023-07-07:
> Hi, I'm sorry but have no way to reproduce your issue, I do not own the radio and on X20, X12 and X10 works fine, it also works on the X18 simulator, so really no clues here

Thank you. Do you maybe remember what crossfire firmware was used in the test? I wonder if this is maybe some sensor issue or something.

yaapu, 2023-07-08:
mhmm you made me realize that GPS data is NOT available via CRSF/ELRS as custom ardupilot telemetry but rather as standard CRSF telemetry, that might be the reason why I don't catch those packets...

jaroszmm, 2023-07-08:
I see. So, anything that can be done/changed from my side?

jaroszmm, 2023-07-22:
A bit of update - I tried both custom and official Arduplane and on both I am having the same issue. Last thing I'm going to try out is R9 to see if this is CRSF issue or not.

jaroszmm, 2023-07-23:
Update #2 
R9mini of F765 with custom Arduplane has no issues displaying GPS data.

So it seems the issue is with CRSF protocol? I tried disablink Mavlink but no luck either.

yaapu, 2023-07-24:
need to replicate thsi because I'm sure it worked before with ELRS 2.x, mhmm

jaroszmm, 2023-07-24:
And one more information.
During one of the crossfire tries somehow I got the bottom GPS data line displaying GPS data however it looked like it got a single data because the GPS coordinates starting with 18.xxxx never updated later. So like somehow it read some GPS data from CRSF but that was by accident. 

I hope you will find the solution to my issue as I'm not sure I want to go R9 on my Arduplane planes :D

jaroszmm, 2023-08-04:
A bit of update.
Arduplane, R9, X18 - no issues, widget is working fine all the time, no lagging etc. Arducustom.

VinceHogg, 2023-10-08:
Same problem here. All was working ok with internal Frsky X20 TX but with CRSF external module, get no GPS data on Yaapu. GPS data is visible on sensors but yaapu screens say 0,0. Other Yaapu telemetry seems ok.

Any update?

yaapu, 2023-11-06:
@VinceHogg would you test a fix?

jaroszmm, 2023-11-06:
I can try it too on my X18 with crsf nano tx if this might help.

jaroszmm, 2023-11-06:
> > I can try it too on my X18 with crsf nano tx if this might help.
> 
> [yaaputelemetry_x18_twlites.zip](https://github.com/yaapu/FrskyTelemetryScript/files/13270505/yaaputelemetry_x18_twlites.zip)

Thank you. I will come back with results soon.

VinceHogg, 2023-11-06:
@yaapu 
Working !
At least on the bench. Not flown yet of course. Thanks for you excellent work.

VinceHogg, 2023-11-06:
Is there a way to zoom with buttons? looks like the 'zoom channel' option is not there on Ethos?

KPruett, 2024-12-26:
Hello Yaapu,

I am trying to get your script running on a new Twin X-lite with Ethos. I have the script installed,  and I'm receiving GPS and attitude information in telemetry for the radio,  but the script says no telemetry. Is there a setting i need to change to let the widget find the telemetry data? 

Secondly,  the top and bottom status bars for ethos are there,  and the widget is moved down,  cutting off the bottom of the widget.  Is this something that can be fixed? 

Thank you!