yaapu/FrskyTelemetryScript #208: Can't compile OTX-ETX_BW-LCD (bw128x64)

Rodrigo-NH, 2023-07-15:
Trying https://github.com/yaapu/FrskyTelemetryScript/releases/tag/OTX-ETX_BW-LCD_ver_2.0.0_beta1 and doesn't work on my Radiomaster TX12. It opens the screens (to get to setup screen is tricky though, I can access it eventualy), but never gets telemetry data (or get some initial frames getting past the "no telemetry" blinking pane, but still showing no telemetry status after all)

Trying to bug hunt but can't have the scripts compiled right. It compiles on EdgeTX simulator but then when trying to run the .luac in the radio I get:

"Syntax error
yaapu7.luac
imcompatible precompiled chunk"

Trying to have the .luac compiled directly in the radio I get:

"Script panic
not enough memory"

I'm able to compile if I start replicating the code (using only code parts in support to the function using crossfireTelemetryPop() 'crossfirePop')
What I see in the radio in my replicated code is execution never get past the https://github.com/yaapu/FrskyTelemetryScript/blob/9e083362190997aad29b915c5a0a835b93bb5bd9/OTX_ETX/bw128x64/SRC/yaapu7.lua#L803 message buffer loop part
It can be a bug (or just something missing in my replication). I'm confused about how msgBuffer is either a local and a global variable and which one are being evaluated in the functions (im just starting with LUA)

I was able to work around this in my replicated code but of course it doesn't mean much if I cannot try the original code.

I think it's worth mentioning:
My TX12 radio got a alternative uC (64k type instead of the usual 128k) because of the chip shortage. It's also some clone uC (CKS CKS32F103C8T6). But I guess the 128k version uses a clone too.
Reference: https://radiomasterrc.freshdesk.com/support/solutions/articles/64000267203

This is not a problem for MULTI-module as they offer pre compiled firmware for the 64k version.

Rodrigo-NH, 2023-07-25:
I think I made a mistake by adding that extra info (unnecessary noise and not well explained also) as the 64k chip is the multi-module one, not the radio one. So probably not related to this issue.

The strange part is it compiles in EdgeTX the PC emulator but gives me the "imcompatible precompiled chunk" when trying to run in the radio.

I need try investigating this more (maybe a issue upstream)

yaapu, 2023-07-25:
I think I've read something about 32 vs 64 numbers in the lua runtime/compiler on edgetx, can't remember where though

Rodrigo-NH, 2023-08-09:
Here probably -> https://discuss.ardupilot.org/t/an-open-source-frsky-telemetry-script-for-the-horus-x10-x12-and-taranis-x9d-x9e-and-qx7-radios/26443/1604
So this is probably https://github.com/EdgeTX/edgetx/issues/1611

Rodrigo-NH, 2023-12-09:
Hi,
I was working for a version for me so I can have telemetry while this blocker exist. Also my impression is the telemetry screens are too heavy for small cpu/ram radios like TX12.

Initially I wondered about forking this project to continue with something different, other direction, but I think it makes sense if ask you first: Do you think it may be a good idea adding another option to this project? Like 'bw128x64-small' or 'bw128x64-alternative' or anything for a name. 
I can try adapting from the original 'bw128x64' and make a pull request later if so.

I started adding basic views, a screen including attitude indicator, Vspeed indicator, home arrow and some readings. And another screen that shows ardupilot messages. It's working in progress as I want to add other indicators like compass and work in the screens layout/functionality.
Thanks Yaapu

[image]

closed 2024-10-10 by yaapu