yaapu/FrskyTelemetryScript #209: no gps despite it showing gps and no other date in screen

martinhally, 2023-07-22:
Hi I have loaded this onto mt X20 and the screen appears as expected but it flashed no gps despite it showing coordinates and has no other data showing in battery etc. sensors are all showing on the configured standard ethos screens. I am using a tandem receiver so as per instructions I can't choose rssi only I have to choose rssi 2.4 or rssi 900. 
 Any idea how I get the date to show and for it to register that it has GPS.

yaapu, 2023-07-24:
Hi,
can you add a screenshot to show your issue in detail.
No GPS, is unrelated to the coordinates, it comes from the GPS status from Ardupilot.
Please also specify which ardupilot version, I don's have any tandem receivers so please be very detailed and patient as well :-)

yaapu, 2023-11-08:
No GPS refers to the "passthrough telemetry status" while coordinates are extracted from sensor data.
It means the script is not receiving telemetry from ardupilot.
looks like an ardupilot configuration issue

closed 2023-11-08 by yaapu