yaapu/FrskyTelemetryScript #210: Baro Altitude issue on ExpressLRS

Cloudfan-official, 2023-07-27:
Does ExpressLRS Support Displaying Altitude From Barometer Sensors?

In my case,when there is no GPS signal,the relative height is always "0"，but the vertical speed can be displayed.

ExpressLRS version is 3.3.0(Both TX and RX)，ArduPilot Version is 4.3.7.

yaapu, 2023-07-27:
Hi, we do not use sensors so if Elrs 3.x supports baro altitude is irrelevant.
vspeed has code to fall back to baro altitude so that explains why it's moving :-)
An ardupilot patch might be needed to send baro altitude when GPS is absent, need to investigate

Cloudfan-official, 2023-07-27:
> Hi, we do not use sensors so if Elrs 3.x supports baro altitude is irrelevant.
> vspeed has code to fall back to baro altitude so that explains why it's moving :-)
> An ardupilot patch might be needed to send baro altitude when GPS is absent, need to investigate

Thanks for your reply.

By the way,Your mean is that we need a patch so that the ArduPilot can send back information about the Altitude from Barometer?

And the patch is currently in development,right?

Cloudfan-official, 2023-07-27:
> Hi, we do not use sensors so if Elrs 3.x supports baro altitude is irrelevant.
> vspeed has code to fall back to baro altitude so that explains why it's moving :-)
> An ardupilot patch might be needed to send baro altitude when GPS is absent, need to investigate

Anyway,there are two things I want to know:
First,whether Yaapu can receive altitude data from barometer on the ExpressLRS system.
Second,if so,how to configure it.

yaapu, 2023-07-27:
Hi, the patch is not in development, nobody wrote it.
My scripts do not use sensors but rather custom telemetry coming from ardupilot so no easy way to support baro altitude.
Last point is current ardupilot CRSF telemetry does not include elrs 3.x custom baro frames since those are "extensions" from the elrs team, I don't think that the ardupilot CRSF maintainer added them

Cloudfan-official, 2023-07-27:
> Hi, the patch is not in development, nobody wrote it.
> My scripts do not use sensors but rather custom telemetry coming from ardupilot so no easy way to support baro altitude.
> Last point is current [...]

So Your mean is that ArduPilot CRSF does not currently provide support for Baro Altitude Telemetry?(I think that is what you mean).

At least,Whether other remote control systems,such as Frsky,support Baro Altitude information on Yaapu?

Thanks.

yaapu, 2023-07-27:
Hi, nope, unfortunately no baro altitude support in ardupilot, as I said it would require a patch to the ardupilot frsky telemetry library which is used to feed both frsky and crsf telemetry for yaapu

closed 2023-07-27 by Cloudfan-official