yaapu/FrskyTelemetryScript #215: No telemetry data with ELRS on Yaapu widget

shark-ru, 2023-09-18:
No telemetry data on Yaapu widget
HW & SW specs and versions:
  Radiomaster TX16S, EDGE-TX v2.8.1
  ELRS Radiomaster - TX, Ranger Micro; RX, RP1-V2. Flashed with version 3.3.0
  Arduplane version 4.3.5
  Yaapu widget version 2.0.0 beta 3
Telemetry sent from FC - Matek H743 to Radiomaster TX16S and updated with selected frequency. However Yaapu seems not getting this data. Debug don't show data exchange.
[image]
[image]
[image]
[image]
[image]
[image]

yaapu, 2023-09-23:
Hi did you enable custom telemetry on Arduplane by setting RC_OPTIONS += 256 ?

shark-ru, 2023-09-25:
[image]

Hi,
Issue solved with this settings. Thank you!

lwan288, 2023-09-25:
> [image]
> 
> Hi, Issue solved with this settings. Thank you!

Hi, Did you resolve your issue by enabling "Use passthrough for CRSF telemetry", right?

I had a similar issue. I can see the telemetry data showing, but not on Yaapu widget.    : ( 

Here are my HW & SW specs and versions:
Radiomaster TX16S, EDGE-TX v2.7.1
TBS Crossfire TX 6.17, XF Nano Div. RX 6.17 
Inav version 6.0
Yaapu widget version 2.0.x dev 
Telemetry sent from FC - Matek H743 to Radiomaster TX16S and updated with selected frequency. 

[image]

[image]

[image]

shark-ru, 2023-09-27:
Hi,
Yes. Passthrough for CRSF solved my issue:)

lwan288, 2023-09-27:
Not sure whether there is a similar option for iNav. It is weird that I can see data in Yaapu Telementry Widget, but still  there is an error saying "no telemetry data" and Horus Mapping Widget doesn't show map.  : (

lwan288, 2023-09-27:
> Hi did you enable custom telemetry on Arduplane by setting RC_OPTIONS += 256 ?

Sorry to disturb you!  I am experiencing a weird problem while the widget is displaying a "no telemetry data" message even though there is data visible on the screen. Do you know what might be the issue with my configuration?  Many thanks in advance!

Here are my HW & SW specs and versions:
Radiomaster TX16S, EDGE-TX v2.7.1
TBS Crossfire TX 6.17, XF Nano Div. RX 6.17 
Inav version 6.0
Yaapu widget version 2.0.x dev 
Telemetry sent from FC - Matek H743 to TBS crossfire TX and then Radiomaster TX16S (Baudrate 400K). 

[image]

[image]

[image]

[image]

[image]

shark-ru, 2023-09-28:
Yaapu is Ardupilot widget. I'm not sure if it'll work with Inav...

lwan288, 2023-09-28:
> Yaapu is Ardupilot widget. I'm not sure if it'll work with Inav...

Thanks! 

I saw previous example by Yaapu showing it works with Inav [https://www.youtube.com/watch?v=tyHzxsb5Tio](url) , but not sure the current status. Or this is simply a result of configuration issue on my end.

yaapu, 2023-09-28:
Hi, it does not work with INAV, my widget requires Ardupilot.
For Inav you should use https://github.com/iNavFlight/OpenTX-Telemetry-Widget