yaapu/FrskyTelemetryScript #216: No telemetry AGL with ELRS on Yaapu widget and CRSF - what about SBUS rx tm now in new  ELRS 3.3 ?

alccrl, 2023-09-25:
Hi,

Elrs 3.3 seems to allow Sbus telemetry with no hw change (no need for hw fport/sport inversion).
Since CRSF Yapau script does not show AGL (altitude above ground (barometr altitude) because CRSF does not contain it, which is Yaapu script and ardupilot/elrs configuration (if any)  in order to use SBUS telemetry ?

I did not find any reference to this on internet, it may be a request for new feature ?
Or does already exist a solution for this ?

Alfredo

yaapu, 2023-10-11:
Ardupilot does not send baro altitude, we send relative home altitude and absolute height above terrain with terrain support.
My script does not use plain CRSF, so I do not read sensor values, I use CRSF to transport a custom payload (passthrough telemetry).
So to recap there's no plan to support baro altitude :-)

alccrl, 2023-10-11:
Thanks,

anyway this topic is a bit foggy to me ...

Vtx arducopter OSD display shows an altitude above ground, so I thought this to be same AGL as SBUS telemetry and used by inav lua script.
May be I (or SBUS naming) confused takeoff altitude, with barometric altitude and altitude respect to ground ....
Summarizing, I get on my rc controller gps altitude, I would prefer takeoff altitude (Inav and Betaflight provide this).
 
Ps:
I tried to configure arducopter terrain feature, but I get errors at arming time, and I have yet to underatand how stable and tested is it, because I cannot take the risk to crash my quad on a mountain if too much relaying on it.
My on board  sd card contains all europe 30m terrain data, but it is disabled in paremeter copter set.

Alfredo

alccrl, 2023-10-11:
Terrain error is
"Waiting for Terrain Data"

yaapu, 2023-10-11:
Hi, I'm sorry but ardupilot setup here is a bit out of scope, whatever you see in my widget comes from ardupilot, I never "generate" alert or messages.
Not everything you see on the vtx can be replicated on my script, telemetry is sent as a restricted set of information, you'll see in my widget only teh restricted set of indormation that "passthrough telemetry provides", so since baro altitude is not sent I really cannot display it :-(
It's not mavlink nor vtx, passthrough is a "light" and very specific telemetry protocol.

closed 2023-11-08 by yaapu