yaapu/FrskyTelemetryScript #219: TX16S widget gets disabled

dxs94, 2023-10-17:
My setup is: TX16S // OTX 2.3.15 // Crossfire latest fw // Arduplane latest beta // Yaapu 2.0.0 beta 3 (latest).
I have 3 pages setup on a single telemetry page (messages / main panel / map).

When flying with the map selected (thus displayued on screen), the widget eventually gets "Disabled" (every flight). As per the Wiki I then went into the widget settings to view the reason for the crash: "CPU Limit" (see pic below). I have no other widget running, but I have a lot of maps in the map directory and of course a bunch of logical switches and functions and mixes in OTX. Is there anything I can do to prevent the widget from crashing (or what is the root cause)?

[image]

yaapu, 2023-11-06:
Hi, I'm afraid I'm pushing it too hard, too much stuff on screen, I bet the altitude graph is causing it to crash

dxs94, 2023-11-06:
I'm also using your standalone mapping widget for Betaflight, which doesn't crash, and I think to recall that one doesn't have the alt graph (not 100% sure though no radio with me).
So could totally make sense.
I'm willing to try to disable the alt graph and report. Would you make a special test version? Or simpler point me in the right direction and I'll edit the .lua files myself directly on the radio.
Thanks!

yaapu, 2023-11-06:
hi try to comment line 80, this line

https://github.com/yaapu/FrskyTelemetryScript/blob/9e083362190997aad29b915c5a0a835b93bb5bd9/OTX_ETX/c480x272/SD/WIDGETS/yaapu/lib/layout_map.lua#L80C3-L80C3

like this

`--local lastY = libs.drawLib.drawGraph("map_alt", 278-4, 167, 120, 48, utils.colors.darkyellow, alt, false, false, nil, nil)`

dxs94, 2023-11-07:
Hi, Thanks for your time!
I commented line 80, now the mapping part of the widget crashes right away at stratup (disabled warning). Debug shows it is not happy with line 86 now (?)
[image]

yaapu, 2023-11-07:
ok, I tested on the sim, we need to comment out the whole section,

-- lcd.drawBitmap(utils.getBitmap("graph_bg_120x48"),278, 167)
--  lcd.setColor(CUSTOM_COLOR,utils.colors.white)
--  lcd.drawText(278+118,167-2,altLabel.." "..unitLabel,SMLSIZE+CUSTOM_COLOR+RIGHT)
--  local lastY = libs.drawLib.drawGraph("map_alt", 278-4, 167, 120, 48, utils.colors.darkyellow, alt, false, false, nil, nil)
--  local altMin = libs.drawLib.getGraphMin("map_alt")
--  local altMax = libs.drawLib.getGraphMax("map_alt")
--  lcd.setColor(CUSTOM_COLOR, WHITE)
--  lcd.drawText(278,167+9,string.format("%d",alt),MIDSIZE+CUSTOM_COLOR)
--  lcd.setColor(CUSTOM_COLOR, lcd.RGB(190,190,190))
--  lcd.drawText(278,167+32,string.format("%d",altMin),SMLSIZE+CUSTOM_COLOR)
--  lcd.drawText(278,167-3,string.format("%d",altMax),SMLSIZE+CUSTOM_COLOR)

dxs94, 2023-11-09:
Runs fine without the graph, waiting on weather for a flight test and will let you know! Thanks!

dxs94, 2024-05-20:
Well finally was eventually able to fly to aircraft again and not a single crash! So for me removing the altitude graph completely fixed the issue.