yaapu/FrskyTelemetryScript #222: Yaapu Ethos widget show "no telemetry" when external module used.

wfield0455, 2023-10-25:
I configured the telemetry widget on my FrSky X20 Pro using the internal Twin module with a TWMX receiver set to F. Port.  In this configuration everything worked perfectly. I then tried the same configuration using an external Twin module in my X20S but the Yaapu widget always shows the big red no telemetry screen. To rule out setup issues beween the 2 transmitters I then installed the Twin module in the external bay of my X20 Pro, bound the receiver and the X20 Pro also shows "No Telemetry" when used with the external module... Both configurations seem to detect the same telemetry sensors but the Yaapu script doesn't seem to be able to detect them using an external module.

yaapu, 2023-10-25:
Hi, I don't have the hardware and have no idea on how to debug this issue but I'd love to

yaapu, 2023-10-25:
I'm trying to compile a list of non working configurations, can you be more precise about yout external twin transmitter model and firmware, same for the receivers please

wfield0455, 2023-10-26:
Configuration

FrSky TWMX Receiver running V1.0.5 firmware option set to F. Port
FrSky Twin Lite Pro module, run-in V1.0.5 firmware.

Model configuration works perfectly using internal module for TW Protocol on X20 Pro. Install the Twin Lite Module in external bay of either the 
X20 Pro or X20S and bind the same TWMX receiver and select F. Port and yaspu//FrSky telemetry script always show No Telemetry.

I plan to do some additional testing soon to see if the ACCESS Protocol has the same issue when used with an external module.

Wayne

wfield0455, 2023-10-26:
Some additional information.
This morning I tried using the FrSky ACCESS Protocol in the hopes that this hardware may be easier for you to obtain. I used an FrSky Archer Plus RS receiver with inverted s. Port output connected to the UART TX Pin, receiver options set for F. Port via the FrSky X20S internal module. In this configuration, everything works properly. When the aircraft is powerd up, yapuu telemetry reports initianlizing, along with other status messages. GPS status is displayed correctly and flight modes and artificial horizon update correctly.
At this point I installed the FrSky Twin Lite Pro module which also supports ACCESS and repeeated the bind command with ACCESS protocol select. All controls work and in this configuration yaapu does NOT report no telelemetry as it did with the TWIN Protocol selected. It also show GPS infomation (coordinated, number of sats, HDOP) correctly. However it does not detect flight mode changes and does not update the artificial horizon as the aircraft is pitched, rolled, etc..
Not sure if this is helpful or simply adds confusion but if you would like me to perform additional tests please advise.

hyperion11, 2023-11-01:
> Hi, I don't have the hardware and have no idea on how to debug this issue but I'd love to

i have this problem too. What kind of debug data i can provide for you to fix this problem?

krzysztofkuczek, 2023-11-05:
Same problem using qczek lrs as an internal module (CRSF) with a radiomaster pocket radio.

axelfenner, 2023-11-05:
Hi, I modified util.lua for working external module R9/TW on ETHOS , Remplace Line 45 on lib/util.lua
Original Line45:  local passthroughSensor = sport.getSensor({appIdStart=0x800, appIdEnd=0x51FF})
Modified Line45:  local passthroughSensor = sport.getSensor({module=1,appIdStart=0x800, appIdEnd=0x51FF})

yaapu, 2023-11-05:
> Hi, I modified util.lua for working external module R9/TW on ETHOS , Remplace Line 45 on lib/util.lua Original Line45: local passthroughSensor = sport.getSensor({appIdStart=0x800, appIdEnd=0x51FF}) [...]

great!, on which radio did you test it?

axelfenner, 2023-11-05:
X10S, R9 ACCST , ETHOS

yak-54, 2023-11-05:
will test on my X12s as well

axelfenner, 2023-11-05:
module = 0, in internal module, module = 1 R9 External module, a configuration tab for choice would be great.

yaapu, 2023-11-05:
> module = 0, in internal module, module = 1 R9 External module, a configuration tab for choice would be great.

sure, as soon as it is confirmed to solve all issues I'll do the mod, thanks again

yak-54, 2023-11-05:
yep works fine on an X12S with R9m and R9 on Flex FCC firmware 👍

yaapu, 2023-11-06:
> Some additional information. This morning I tried using the FrSky ACCESS Protocol in the hopes that this hardware may be easier for you to obtain. I used an FrSky Archer Plus RS receiver with [...]

@wfield0455 if you could try as well would be great!

yaapu, 2023-11-06:
[yaaputelemetry_x20.zip](https://github.com/yaapu/FrskyTelemetryScript/files/13270266/yaaputelemetry_x20.zip)
[yaaputelemetry_x10_x12.zip](https://github.com/yaapu/FrskyTelemetryScript/files/13270267/yaaputelemetry_x10_x12.zip)
[yaaputelemetry_x18_twlites.zip](https://github.com/yaapu/FrskyTelemetryScript/files/13270269/yaaputelemetry_x18_twlites.zip)

@yak-54 @axelfenner @hyperion11 @wfield0455 would you guys please test this fix?
just unzip over the same files in the [SD ROOT]/scripts/yaaputelemetry directory

axelfenner, 2023-11-06:
On X10S not diplay config option TelemetrySourse. I copy from x20s on main lua this configure function and select 3, but no telemetry msg.

yaapu, 2023-11-06:
> On X10S not diplay config option TelemetrySource. I copy from x20s on main lua this configure function and select 3, but no telemetry msg.

auch, you're right I missed a changed file, sorry

yaapu, 2023-11-06:
[yaaputelemetry_x18_twxlite.zip](https://github.com/yaapu/FrskyTelemetryScript/files/13271842/yaaputelemetry_x18_twxlite.zip)
[yaaputelemetry_x10_x12.zip](https://github.com/yaapu/FrskyTelemetryScript/files/13271855/yaaputelemetry_x10_x12.zip)
[yaaputelemetry_x18_twxlite.zip](https://github.com/yaapu/FrskyTelemetryScript/files/13271861/yaaputelemetry_x18_twxlite.zip)

let's hope I got it right this time :-)

bsongis, 2023-11-07:
May I close this issue then?
https://github.com/FrSkyRC/ETHOS-Feedback-Community/issues/2245

yak-54, 2023-11-07:
> May I close this issue then? [FrSkyRC/ETHOS-Feedback-Community#2245](https://github.com/FrSkyRC/ETHOS-Feedback-Community/issues/2245)

i will now

yak-54, 2023-11-07:
> [yaaputelemetry_x18_twxlite.zip](https://github.com/yaapu/FrskyTelemetryScript/files/13271842/yaaputelemetry_x18_twxlite.zip) [...]

yes you have tested on X12S ETHOS 1.4.13 with R9M and R9 on flex firmware and IXJ with R8 Pro in the module bay as well

hyperion11, 2023-11-07:
@yaapu test fix on twin xlite pro with R-XSR access recever and R9 external module. Its working. Thanks!

wfield0455, 2023-11-07:
Not sure if I’m doing something wrong but it still doesn’t work for me.

I extracted the X20 tip file, coped all the files to the appropriate directories on my transmitter, installed a TwinLitePro module in my X20S, selected external module via config widget for yap screen and yap screen still displays no telemetry.

Wayne

wfield0455, 2023-11-07:
Also, I not using SBUS / S.Port, I’ve configured my receiver (Archer Plus RS) via set command to F.Port if that makes any difference.

wfield0455, 2023-11-07:
Sorry, I forgot to rediscover telemetry when switching to external module. It’s working on my x20 for both internal and external moduleSent from my iPhone

yaapu, 2023-11-08:
@axelfenner thanks once more

closed 2023-11-08 by yaapu