yaapu/FrskyTelemetryScript #223: Support BW version of script for radios without Menu button

lesykm, 2023-12-17:
On a BW radios without menu button it is impossible to call for config page of the script.
You may consider adding EVT_SYS_LONG or similar here: https://github.com/yaapu/FrskyTelemetryScript/blob/ca15020935b6dee29940de449a49ae542dd49b48/OTX_ETX/bw128x64/SRC/yaapu7.lua#L1597 to support such radios.

yaapu, 2024-01-10:
> On a BW radios without menu button it is impossible to call for config page of the script. You may consider adding EVT_SYS_LONG or similar here: [...]

I need to be able to simulate the radio, which one do you refer to?

yaapu, 2024-01-10:
> Please consider adding this. The MT12 doesn't have a menu button.

official edgetx does not support the MT12, which version are you running? I need to be able to simulate it...

frodus17, 2024-01-10:
> > Please consider adding this. The MT12 doesn't have a menu button.
> 
> official edgetx does not support the MT12, which version are you running? I need to be able to simulate it...

It runs EdgeTX 2.10, but it's a version that Radiomaster compiled or EdgeTX may have helped compile for them. I have the firmware if you can simulate with that.

yaapu, 2024-01-10:
> It runs EdgeTX 2.10, but it's a version that Radiomaster compiled or EdgeTX may have helped compile for them. I have the firmware if you can simulate with that.

nope I need a windows or linux build of the simulator
Which button do you use to enter edgetx menus?

frodus17, 2024-01-11:
[image]
This is what I have on my radio. No button I press can get me a menu screen. 

Is there maybe a workaround to instead use a text file on SD card to configure? Yml? I don't need access to the menu after setup. Not sure where the configuration is stored once you change in the script.

yaapu, 2024-01-11:
Yes, you can try to edit config file, check in the /MODELS/yaapu folder for a .cfg file, change CRSF:0, to CRSF:1 or vice versa, should work (you need to power cycle)

frodus17, 2024-01-11:
There is only yaapudev.cfg in the MODELS/yaapu/ directory. It does not contain the CRSF:0 (or any context that reads CRSF) in the file. Does the config file save after configuration? If so, I can't configure, so likely isn't getting generated. 

Do you have a default one?

robertlong13, 2024-02-16:
If you want an example that you can simulate, the RadioMaster Boxer, Zorro, TX12 Mark II, all have the same issue. Basically all the RadioMaster B/W radios.

I do have a workaround though (I'm not sure how it works):
- Press the telem button to open the HUD
- Long press RTN to exit
- Press the telem button again, config menu is now open.
- Short press RTN to get back to the HUD

frodus17, 2024-02-16:
> If you want an example that you can simulate, the RadioMaster Boxer, Zorro, TX12 Mark II, all have the same issue. Basically all the RadioMaster B/W radios.
>
> I do have a workaround though (I'm not [...]

Hmm, I tried that but couldn't get it working on the MT12. It never shows the config menu. What radio are you using?

Maybe it'd be easier to define the configuration file naming, location and variables to change so we can manually change to get it working. It's been about a month and unfortunately I've tried everything I could think of.

yaapu, 2024-09-23:
Hi, please check if this helps

https://github.com/yaapu/FrskyTelemetryScript/issues/196

closed 2024-09-23 by yaapu