yaapu/FrskyTelemetryScript #226: Yaapu script broken on Ethos 1.5

iamwinq, 2024-02-06:
Ethos 1.5 has changed timers that led to yaapu script not useable... will this be updated in the revision of the Yaapu for ETHOS?

[image]

yak-54, 2024-02-06:
Ethos brakes lot Lua scripts on 1.5.0 test firmware 
 i would say it stay like that till they release 1.5.0 as its just an beta firmware they may change it again before  its released

yaapu, 2024-02-28:
try this in the meantime
https://discuss.ardupilot.org/uploads/short-url/pueRO57725BrEAUXi8FsUa6aCrv.zip

JoshWelsh, 2024-03-10:
> try this in the meantime https://discuss.ardupilot.org/uploads/short-url/pueRO57725BrEAUXi8FsUa6aCrv.zip

This does seem to work for me.. I installed 1.5.2 and then this patch and so far on the bench everything seems good again.  Once again sir, you are the man!

iamwinq, 2024-03-16:
> try this in the meantime https://discuss.ardupilot.org/uploads/short-url/pueRO57725BrEAUXi8FsUa6aCrv.zip

i have tested it as well, it works perfectly! Thank you Alex :)

yaapu, 2024-03-30:
fixed by latest version 1.1.0 on github

closed 2024-03-30 by yaapu