yaapu/FrskyTelemetryScript #227: BW128x64 radios (Taranis x9lite) have no config script and EdgeTx model copy function does not copy config file

alccrl, 2024-03-01:
Hi,
trying to put into work a new (copied) model on tranmitter with EdgeTx and Elrs,
copied from a working model,
after troubleshooting ("NO TELEMETRY") on the new model, I discovered that
Yaapu Config script file in SD content for 128x64B&W radios was missing.
and that EdgeTx copy model functionn did not create a config file for the copied model.

Hence ...
I manually copied the model script file working into a new model file, changed the name to the new model name and
now script for new (copied) model works.

Version : Have a lok at dev branch on github.
Other details on request.

Alfredo

alccrl, 2024-03-01:
Addendum:
I mean I copied yaapu script model file, not EdgeTx model file.

Alfredo

menschel, 2024-03-30:
I had the same issue and found help.

https://discuss.ardupilot.org/t/help-needed-with-yaapu-lua-script-and-telemetry-setup/113013/6

It comes down to being on the yaapu telemetry page and pushing RTN followed by TELE to access the config.
Scroll to the bottom, select CRSF  and reboot the radio.
A high telemetry ratio 1/4 is needed. I had RC failsafe issues with 1/32.

alccrl, 2024-03-30:
Your suggested link explains how to configure ELRS on Yaapu from scratch.

Problems arise when You copy or create another (elrs) model on transmitter.
Moreover ..., 
I did not succeded in  using two elrs 900 receivers on two different quads with same transmitter.
I searched a way to configure different pass phrases for each quad (two different models on same transmitter), but I had to use same passw for both... this topic is not very clear on elrs site, so I stopped searching a solution.

Alfredo

menschel, 2024-03-31:
OK, then I misunderstood that part.

I just tested copying models on the tx12mk2 with the yaapu script.

For me, it works. I can copy the model, switch to the copy of the model and it picks up telemetry right away.

I don't have a second receiver right now.

closed 2024-10-10 by yaapu