yaapu/FrskyTelemetryScript #228: airspeed and groundspeed indication with ardupilot FW

kostaslbc, 2024-03-05:
Is it posssible to display both airspeed and groundspeed with arduplane? Airspeed indication only for long range flights is not safe.
Thanks.

rscarawa, 2024-04-05:
Is this not already done?  I thought they show both GPS speed (same as ground speed) along with airspeed on the left side of the HUD.  
[image]

[image]

I thought the green number was GPS speed while the number under it was airspeed.  Am I incorrect?  Are the HUD elements documented anywhere?

iamwinq, 2024-05-07:
it's working on mine though

yaapu, 2024-09-20:
https://github.com/yaapu/FrskyTelemetryScript/wiki/airspeed-vs-groundspeed

closed 2024-09-20 by yaapu