yaapu/FrskyTelemetryScript #229: Enabling CRSF passthrough support on Radiomaster Pocket?

hhackbarth, 2024-03-25:
How to enable CRSF support in the Yaapu telemetry settings on a Radiomaster Pocket?
On the Boxer and the Zorro, the sequence TELE -> RTN -> TELE brings you into that configuration menu, but on the Pocket, this does'nt work. I found no other way to get into the yaapu configuration settings (using yaapu 2.0.0 RC3)

iamwinq, 2024-03-31:
Can't access yaapu config menu as well...any fix for this?

hhackbarth, 2024-03-31:
_At the moment, I this workaround works for me:_
When you have another remote controller with same Yaapu scripts version and where you are able to enter the configuration screen, you can copy the generated config file (in SD card folder models/yaapu/<modelname>.cfg) into the same location on the SD card of the Radiomaster pocket.

Looking forward to a fix...

iamwinq, 2024-04-01:
Can you share your config file for the pocket?

hhackbarth, 2024-04-02:
> Can you share your config file for the pocket?

My config file currently looks like this:
L1:1,V1:320,V2:300,B1:0,B2:0,CC:0,CC2:0,BC:1,VS:1,S1:1,S2:3,VIBR:1,T1:0,A1:0,A2:0,D1:0,T2:10,RM:0,HSPD:1,VSPD:1,CPANE:1,RPANE:1,LPANE:1,AVIEW:1,PX4:1,CRSF:2

It's stored in the folder MODELS/yaapu under the name <ModelName>.cfg, where ModelName is the name of the model, that you are currently using.

Please note, that I also changed the voltages for warning (3.2V) and critical (3.0V)to match the properties of LiIon cells.

iamwinq, 2024-05-07:
I tried your config but still it shows "no telemetry"...

iamwinq, 2024-05-07:
oh! got it working, I needed to rename the .cfg file to whatever name of my model is

Hwurzburg, 2024-06-06:
here is a new yaapu7.lua that will bring up the config screen with a long press on the MDL key of the Pocket....go to SCRIPTS/TELEM and delete the yaapu7.luac and unzip and copy this one there.....the telem key will bring up the main telem screen, then long press the MDL key to get to the config utility
[yaapu7.zip](https://github.com/user-attachments/files/15686489/yaapu7.zip)

iamwinq, 2024-06-18:
> here is a new yaapu7.lua that will bring up the config screen with a long press on the MDL key of the Pocket....go to SCRIPTS/TELEM and delete the yaapu7.luac and unzip and copy this one there....it will compile to a .luac on first access....the telem key will bring up the main telem screen like it does now, then long press the MDL key to get to the config utility....enjoy [yaapu7.zip](https://github.com/user-attachments/files/15686489/yaapu7.zip)
> 
> @yaapu Alex, simple additional OR on the config screen conditional to enable it for Pocket

THank you! this worked flawlessly

gwvr, 2024-07-14:
I filed https://github.com/yaapu/FrskyTelemetryScript/issues/239 and have just seen this issue.

I've tried copying the [config above](https://github.com/yaapu/FrskyTelemetryScript/issues/229#issuecomment-2031263434) to `MODELS/yaapu/[ModelName].cfg`, but that seems to have had no effect.

I've also tried removing `yaapu7.luac` and copying `yaapu7.lua` to `SCRIPTS/TELEM`, as suggested [above](https://github.com/yaapu/FrskyTelemetryScript/issues/229#issuecomment-2153418035), but that causes my Pocket to continuously reboot (loop the) loop.

yaapu, 2024-09-20:
Hi please try this script, it's a standalone one, copy it in the /scripts/tools folder and run it from the tools menu (same place where you have Yaapu Debug) and let me know.
[Yaapu Config.zip](https://github.com/user-attachments/files/17074259/Yaapu.Config.zip)
You'll have to restart the telemetry script to pick up the changes

yaapu, 2024-09-23:
please see here

https://github.com/yaapu/FrskyTelemetryScript/issues/196

closed 2024-09-23 by yaapu