yaapu/FrskyTelemetryScript #231: Ethos 1.5.4 X20 pro goes into emergency mode with screen Title on

yak-54, 2024-03-30:
Ethos 1.5.4 X20 pro goes into emergency mode with screen Title on

ethos 1.5.4 when i go through my screens this happens when i get to yaapu screen with the Title on
[image]

yak-54, 2024-03-30:
ok worked it out with 1.5.4 when you add telemetry page remember to turn off screen Titles in your lua scripts if you leave it on go to that page the radio will crash

so looks like ethos has issues with screen Titles

iamwinq, 2024-03-30:
let me try your suggestion. I ran into the same EM problem in the field today. i was forced to fly without the yaapu... didnt have problems in flight but it was quite inconvenient without the Yaapu

iamwinq, 2024-03-30:
it worked! thanks buddy!

yaapu, 2024-03-30:
fixed in latest version 1.1.0 on github

closed 2024-03-30 by yaapu

yak-54, 2024-03-30:
thanks Alex

iamwinq, 2024-03-30:
> fixed in latest version 1.1.0 on github

thanks!!!