yaapu/FrskyTelemetryScript #232: Yaapu 1.5 no sound in Ethos 1.5.2

rotemshf, 2024-03-31:
Hello,
After updating the Yaapu to 1.5 there is no sound, no beeps, no announcements.

Please advise.

Thank you very much for any help.
Rotem

rscarawa, 2024-04-04:
I have sound in the widget.  I have an X20S on Ethos 1.5.2.  I did an upgrade to the latest Lua files from late March, 2024.

rotemshf, 2024-04-04:
Hello rscarawa,
How can I upgrade the LUA files?

Thank you
Rotem

rscarawa, 2024-04-05:
I manually picked them out from the https://github.com/yaapu/FrskyTelemetryScript/tree/master/ETHOS/c800x480/SD/scripts/yaaputelemetry path.  You should see that some files were updated in the past week.  Get those at a minimum.  Then make sure your audio files are in the D:\scripts\yaaputelemetry\audio path on your card.  There is a subdir for en (english), de (german), fr (french), and it (Italian).  Make sure there are audio files in there too.  I did all of this by removing the card from the tx and plugging it into my pc.  THe files will need to be downloaded from github and moved to your flash card.  Let me know if that works for you.

rscarawa, 2024-04-05:
Does your tx play audio other than what you expect from Lua?

Do you have sound enabled within the widget configuration?  The configuration for the widget is under screen configurations.  2/3 down in the list, there is 'Disable All sounds'.  Set this to 'No' if not already set there.

Is your file/directory structure the same as mine?  I am only showing directories for brevity.  Some of the directories are not relevant here.  While I do have some audio files under the directories below:
/audio 
/audio/en/
/audio/en/us
/audio/en/us/system

I think the audio that Yaapu cares about is under:
/scripts/yaaputelementry/audio
/scripts/yaaputelementry/audio/en

PS D:\> get-childitem -Directory -recurse |tree
Folder PATH listing
Volume serial number is 0403-0201
D:.
├───bitmaps
│   ├───models
│   ├───gps
│   └───yaapu
│       └───maps
│           ├───GoogleSatelliteMap
│           │   ├───10
│           │   │   └───619
│  ...
│           └───sat_tiles
│               ├───-1
│               │   ├───234
│               │   │   ├───66
│               │   │   │   └───154
│               │   │....
├───audio
│   ├───en
│   │   ├───system
│   │   └───us
│   │       └───system
│   ├───de
│   ├───fr
│   └───it
├───models
│   └───Model Backup
├───logs
├───firmware
│   └───Receiver
└───scripts
    ├───yaaputelemetry
    │   ├───bitmaps
    │   ├───lib
    │   └───audio
    │       ├───de
    │       ├───en
    │       ├───fr
    │       └───it
    └───yaapudebug

Hope this helps

yaapu, 2024-10-10:
stale, closing it

closed 2024-10-10 by yaapu