yaapu/FrskyTelemetryScript #233: Map not showing on screen

rscarawa, 2024-04-05:
When using the map feature, it seems to default to the 'no map' tile.  Anyone else seeing this issue?  How can I tell what tiles should be displayed for my location?  Is there a tool I can plug my gps coordinates into to get a directory and tile file that I can use to verify that the map file exists?

I am on X20S with Ethos 1.5.2.  

Thx

iamwinq, 2024-04-11:
My ethos is on 1.5.4, and I can't use the yaapu maaping script with it. I had it working before, now it seems that the yaapu script can't get h GPS coordinates/GPS data. Main screen also shows lat/long 00000000 even if I got GPS lock.map tiles are not loading on the yaapu mapping script as well

iamwinq, 2024-04-12:
> When using the map feature, it seems to default to the 'no map' tile. Anyone else seeing this issue? How can I tell what tiles should be displayed for my location? Is there a tool I can plug my gps [...]

Are you using crsf?

yaapu, 2024-10-10:
this looks stale, closing

closed 2024-10-10 by yaapu