yaapu/FrskyTelemetryScript #235: No telemetry in TX16S mkII after update to EdgeTX 2.10.0

nspector8, 2024-05-15:
After updating TX16S mkII to EdgeTX 2.10.0, loading all default SD contents for 2.10.0, and reinstalling yaapu data following "Installation on Horus radios". no telemetry is being detected by the radio or shown by yaapu, however the RFD TXMOD in the module bay is still able to send all telemetry to Mission Planner over WIFI via TCP.

Also noticed power light is now green rather than blue, but this may be a radio firmware change I'm unaware of.

The yaapu widgets display correctly in the radio but no telemetry.

Rolling firmware back to EdgeTX "Providence" v2.9.4 results in telemetry restored into yaapu.

Any assistance is greatly appreciated.

keithluneau, 2024-05-22:
I see the same problem here with a TX16S on EdgeTX 2.10.0, using Yaapu version 2.0.0 beta2. I'm using a Dragonlink radio set up to send telemetry into the radio, as well as out to Mission Planner over WiFi/UDP. I have no telemetry showing in Yaapu, and no sensors are detected on the model settings telemetry tab. Everything was working fine before updating EdgeTX to 2.10.0

Squirlier, 2024-05-23:
Hello, I had a similar Problem. maybe due to copy issues on sd card.(which i corrected)
( Horus X10S )
I still had no telemetrie in yaapuu. (only in yaapuu)
In Yapuu config i paused it. (first option). > Widget says "..paused.."
then reenabled / unpaused in yaapuu config. > all telemetrie data ist showing and sensor scan does work also for the missing passtrough yaapuu sensors.
so: pause and unpause maybe was the trick for me.

keithluneau, 2024-05-23:
I just got one of my old radios out, it's a TX16S that still has EdgeTX 2.9.2 on it. Everything worked there as expected. I rolled back my primary radio (TX16S Mark II Max) to EdgeTX 2.9.4 and everything works again. It seems something in 2.10.0 broke telemetry.

For the sake of complete info, I'm using a Dragonlink 915MHz transmitter module with it's internal BT/WiFi module flashed with Mav2PT, configured to push S.Port telemetry out to the radio through the module bay at 57600 baud. This all works perfect with full Ardupilot telemetry in Yaapu prior to EdgeTX 2.10.0

1280MHz, 2024-05-28:
Running Yaapu 2.0.x dev (6cccb30). 

Just updated to EdgeTX 2.10.0, also having the same issue with DL/mavlink telemetry being sent to X10s/Heltec32 WIFI. 

Yappu Debug shows no data. I am able to connect to MP with full telemetry. 

Deleted old Yaapu folders/files and replaced with no luck. 

Looks like I will also be rolling back to Edge Providence.

PhilColbert, 2024-06-30:
Same here, TX16S - Edge TX 2.10.1

Can see all the new telemetry sensors, but script reports no telemetry data?

dbpprt, 2024-07-07:
Same here, Radiomaster Boxer.

Amigache, 2024-08-01:
I am having the same problem as the rest of the users. I am using DL with Mav2PT, and after updating edgetx to 2.10.0, now I do not receive telemetry on the station and the script, of course, does not show anything.

dronesalot, 2024-12-02:
Same problem here on Edge 2.10.4. I can see the telemetry under the model, CRSF/Diversity/PX4, but it always says no telemetry and the debug screens show 0hz. Has anyone managed to resolve this?

nspector8, 2024-12-02:
It appears that the EdgeTX team was able to fix this.

Please refer to issue [#5064](https://github.com/EdgeTX/edgetx/issues/5064#issuecomment-2509660130) and fix [#5689](https://github.com/EdgeTX/edgetx/pull/5689) on the EdgeTX github.

External modules will have drop-down sub type boxes for selecting telemetry type for use with PPM and SBUS and should be included on an upcoming release of EdgeTX firmware.

I'll close this issue.

closed 2024-12-02 by nspector8