yaapu/FrskyTelemetryScript #239: Unable to enter configuration menu on Radiomaster Pocket

gwvr, 2024-07-14:
My Radiomaster Pocket is running the "RM Factory" version of Edgetx, with ELRS. I haven't tried updating to EdgeTx 2.10 because I've seen https://github.com/yaapu/FrskyTelemetryScript/issues/235.

The Tx is receiving telemetry from Ardupilot and I have it set up to give an audible Rx battery low voltage warning.

I've just installed the Yaapu script and it displays but shows "no telemetry". I am unable to enter the Yaapu configuration menu with TELE -> RTN -> TELE. I can bring up the Yaapu debug screens.

How do I access the configuration menu?

gwvr, 2024-07-14:
I was unaware that CRSF passthrough needed to be enabled on the flight controller for Yaapu telemetry (if this is explicit in the Yaapu docs, I didn't see it).
This [thread/post](https://discuss.ardupilot.org/t/yaapu-telemetry-on-opentx-elrs-frsky-horus-x10/77844/18) gave me the information I needed to configure arducopter and enable Yaapu telemetry. Telemetry updates were slow initially, much improved by changing the telemetry ratio.

I still can't access the configuration menu.

yaapu, 2024-09-20:
Hi please try thsi script, it's a standalone one, copy it in the /scripts/tools folder and run it from the tools menu (same place where you have Yaapu Debug) and let me know.
[Yaapu Config.zip](https://github.com/user-attachments/files/17074222/Yaapu.Config.zip)

closed 2024-09-20 by yaapu

gwvr, 2024-09-24:
@yaapu that works for me, many thanks!