yaapu/FrskyTelemetryScript #241: GPS position on FrSky TWIN X Lite S is incorrect

stefan1messmer, 2024-07-28:
The Yaapu telemetry script displays wrong GPS position on FrSky TWIN X Lite S. The displayed latitude and the longitude values are identical and depend from the actual position. The GPS position in the telemetry page, however, is correct.
ETHOS Firmware Version 1.5.11 EU 
Bootlader Version 1.14.16
Module Version 1.0.6

Ardupilot version 4.4.x running on Holibro Pix32V5

yak-54, 2024-07-29:
ethos 1.5.11 broke this
 they have change something in the way it handles gps
[image]

yak-54, 2024-07-29:
[image]

yaapu, 2024-08-29:
Hi, this is how it should have (ever) been :-)

    gpsData.lat = system.getSource({name=status.conf.gpsSource:name(),options=OPTION_LATITUDE}):value()
    gpsData.lon = system.getSource({name=status.conf.gpsSource:name(),options=OPTION_LONGITUDE}):value()

thanks for your help @stefan1messmer

closed 2024-08-29 by yaapu

stefan1messmer, 2024-08-29:
I found the same solution last weekend, but could not evaluate it carefully up to now.

Best regards
Stefan